NASA-Tensegrity-Robotics-Toolkit/NTRTsim #119: Contact Cables Cannot Wrap

brtietz, 2015-01-02:
If a tgBulletContactSpringCable gets wrapped a full turn around an object, it immediately loses all of its anchors. See muscleNPTests demo in my (btietz) dev folder. This is likely due to the anchor sorting scheme in the anchorCompare struct. While not a major problem for tensegrity structures, a more sophisticated sorting scheme would allow for wrapped cables to simulate motor spools and other aspects.

brtietz, 2015-01-07:
Upon further investigation, the cables can wrap if one does not delete based on the results of the anchor contact testing (see lines 518 and 564 of tgBulletContactSpringCable.cpp as of commit 83f492c4158d6b562ef9529c56b3f38d9bb422a4). However, then anchors never lose contact with their respective objects based on sliding off. If we can determine a scheme for lost contact that does not involve btPersistentManifolds, we should be able to make progress on wrapped cables.

vsunspiral, 2015-01-07:
> Upon further investigation, the cables can wrap if one does not delete based on the results of the anchor contact testing (see lines 518 and 564 of tgBulletContactSpringCable.cpp as of commit 83f492c). However, then anchors never lose contact with their respective objects based on sliding off. If we can determine a scheme for lost contact that does not involve btPersistentManifolds, we should be able to make progress on wrapped cables.

this might be where dijkstra's algorithm helps out.  basically see if the straight line path between the prior anchor and next node after the last attached anchor are in free space.  we would need a 3d version of dijkstras (it is normally 2d) which I'm sure exists (and has some other name).

http://en.wikipedia.org/wiki/Dijkstra%27s_algorithm

vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.