NASA-Tensegrity-Robotics-Toolkit/NTRTsim #126: -c option for setup scripts

brtietz, 2015-01-15:
One possible enhancement after we transition the existing setup scripts to Python would be to give users the option to clean out specific sections of their env directory i.e. delete bullet without touching anything else. This would make versioning easier, as users would not have to rebuild all of env every time it changes if they don't know how to manually clean out packages. It would also make the job of version upgrade scripts easier to have these functions.

Thoughts?

PerryBhandal, 2015-01-15:
I'm glad you brought that up, Brian. I've been working on the Python setup scripts and I had a similar idea.

Right now I have a BaseInstaller script that every installer should inherit from. It has three methods: isInstalled, install and uninstall.

While we could make an uninstall method for the existing env structure, an easier way might be to start putting each package in a separate folder. So your env would look something like this:

*downloads*
*bullet*
  src
  li
  include
*boost*
  src
  lib
  include

That would make the uninstall script pretty easy: just delete the entire directory.

I can't see any real disadvantage to this structure over our existing structure. Can anyone see any shortcomings/problems that I haven't?

PerryBhandal, 2015-01-15:
Unfortunately my indentation didn't work in that post, hopefully the layout is fairly clear (every package we install has a src, lib and include directory).

brtietz, 2015-01-15:
Perry,

I like that structure generally for the purposes you outline. One note: it
will require changes to src/inc.CmakeBullet.txt to get the new include
paths correct, but that's not a major hurdle.

Brian

PerryBhandal, 2015-01-15:
Great, thanks for the heads up.

I'm hoping most of the Python scripts will be done by the end of the weekend. I'll post another message once it's ready for testing.