NASA-Tensegrity-Robotics-Toolkit/NTRTsim #128: Python scripts: do we need to retain the ability to install to /usr/lib?

PerryBhandal, 2015-01-18:
Does anyone install NTRT and its dependencies into /usr/lib? Or does everyone use an env based build?

As I migrate the Bash scripts to Python, I'd like to standardize our env structure so it's more consistent across packages. For example, I want to remove the ability to specify which directory gmocktest gets extracted to, which it gets built in, etc. My preferred approach would be to give each package a folder in env with three subdirectories: src, lib, include. 

So, I ask: 

1) Does anyone need the ability to install to /usr/lib?

2) Does anyone need the ability to specify which folder in env gmocktest is extracted to, which it's built in, etc? Or is it fine if we force all packages into an "env/package_name/src,lib,include" arrangement?

3) Can anyone think of any reason these changes shouldn't be made?

PerryBhandal, 2015-01-18:
I suppose I should pitch a broader question as well: is there anything anyone dislikes about our current setup and/or build process? Are there any additional capabilities you want to see?

brtietz, 2015-01-18:
One thing that we should retain is the ability to install boost from a
package. This saves a lot of time and provides a nice backup - the env
install failed for one of my students but packages worked. This could be a
lot simpler than our current setup of course.

PerryBhandal, 2015-01-18:
Thanks for the feedback Brian. Would the following suffice?

** gmocktest.conf (similar cvar would be in every package's conf) **

GMOCKTEST_INSTALL_TO_ENV=false

** general.conf **

ADDITIONAL_LIB_PATHS="/usr/lib;/home/perry/someotherlibpath"

ADDITIONAL_INCLUDE_PATHS="/usr/include;/home/perry/someotherincludepath"

brtietz, 2015-01-18:
That would suit my needs. Thanks!

ryan1248, 2015-01-19:
Hi Perry,

Maybe let's have a call to discuss sometime this week? I'd like to get a
better understanding of the approach you're outlining and make sure we're
taking into account the build dependencies (CMake, etc.). Also we may want
to discuss having src, lib, and include under env, with each package
placing its relevant files in these. Thoughts?

-Ryan

PerryBhandal, 2015-01-19:
Hi Ryan,
I'm generally free from 5 to 9PM EST every day of the week excluding Wednesday. If that works for you, let me know. If not, I'm often around during the day between lectures/labs -- it's best to catch me on Google Hangouts in that case.

The main motivation for the env/<package_name>/ approach is to ease package management. Want to remove bullet? Delete env/bullet. Of course, I can easily modify the code to use an env/lib, env/src, env/include layout if it offers advantages.

Perry

PerryBhandal, 2015-01-20:
Here are the conclusions from my discussion with Ryan this evening:

1) We'll likely use the env/package_name/src,lib,include approach.

2) .conf files are going to become .ini files. Their format will be changing slightly to be compatible with Python's ConfigParser module.

3) We'll use a /var/log style approach to logging rather than what I communicated in my e-mail to the list earlier today. So setup.py will write to base_dir/log/setup.log, build.py will write to base_dir/log/build.log.