NASA-Tensegrity-Robotics-Toolkit/NTRTsim #132: Best location for Python scripts for manipulation of learning results

brtietz, 2015-01-25:
Alex, Steve, and I are looking for a place to dump useful Python scripts that either (a) structure learning runs for useful behavior and/or (b) manipulate the data files associate with learning in a useful fashion. I know Steve has some buried in his dev folder, but this does not strike me as a useful place for new users to find and use them. I think moving them to some place like src/learning/scripts might make them obvious, but not necessarily useful due to file paths. Other thoughts on the best place to put these? Its safe to assume the data being manipulated are in the resources folder.

vsunspiral, 2015-01-26:
It is reasonable to ask people to add 
src/learning/scripts
To their path environment if they want to use the scripts. 

Would the scripts be called from within an executable? 

Or from the command line?

--------------------------------------------
Vytas SunSpiral    
PI - Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to arrive safely at death.

brtietz, 2015-01-26:
The scripts we are using are outside of NTRT simulations, so they are
called from the command line.

So far the path issue has been the reverse - the scripts need to be able to
handle a file path such that they can manipulate files in their own
directory. Shouldn't be too hard to fix.
I'm mostly trying to minimize the usage of ../ when calling the scripts.

vsunspiral, 2015-01-26:
Put a symlink in the scripts dir to the rescources dir???

--------------------------------------------
Vytas SunSpiral    
PI - Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to arrive safely at death.

PerryBhandal, 2015-01-29:
I've put my scripts inside the python_scripts folder in bin/ (see python_scripts branch).

Those scripts are mainly concerned with setup/building, but they offer some useful helpers for navigating paths relative to NTRTsim's base dir, among other things.

If desired, I could modify the structure a bit so there is a new top-level folder where we can separate setup and install scripts from your learning scripts. We could also subdivide the supporting classes based on what they're used for, with a single common folder where helpers/libs that are used globally are stored.

PerryBhandal, 2015-01-29:
An alternate approach if it's preferred to keep your scripts out of bin (that doesn't seem like an intuitive place to put learning scripts) I could add something so that during build my helper python scripts are available in path. That way you can benefit from the helper scripts for directory traversal (relative to base dir) without having to put your scripts in bin.

ryan1248, 2015-02-01:
What about adding a 'scripts' directory at the top level? That's fairly
common in a lot of the popular frameworks so it might be more familiar to
new folks.

Thoughts?

-Ryan

PerryBhandal, 2015-02-01:
I like the idea -- no objection here!

brtietz, 2015-03-20:
Closed by #141

closed 2015-03-20 by brtietz