NASA-Tensegrity-Robotics-Toolkit/NTRTsim #134: Ditch setup.sh/build.sh and move to a single script parameter approach?

PerryBhandal, 2015-02-13:
In the very near future I'll have everything completed to move us from Bash scripts for setup to Python. I discuss the motivations for this change in issue #128.

What I've done so far is I've replaced the existing shell scripts, but I've left build.sh and setup.sh in place. The only thing is those two scripts now just call a Python script rather than doing setup/building on their own.

I propose that we ditch those two scripts as well, and instead have all NTRT operations (starting a build, running setup, and a few others I'll discuss shortly) called from a single file called runner.py.

If we make that change, building and setup will be done as follows

python runner.py setup <-- runs setup
python runner.py build <-- does a build.

Any args that need to be passed to the script can be included after the command, e.g. python runner.py setup --verbose would pass a single argument (--verbose) to setup.py

When I introduce the Python changes, I'll also be adding in code cleanup tools. We'll have three tools:

Uncrustify
Flint
A custom tool which checks project specific requirements (copyright and doxygen header, for instance).

We can use each of those commands either against the whole project, just on a single dir, a single file, etc.

If we use a shell script for each of those tasks as well, we're going to have a lot of scripts. We'll also need to have different scripts for Windows once I return Windows support, so we'll have setup.sh and setup.bat, etc.

If we go with the runner.py approach, we have a single script that is the entry point to all project operations. It's sort of like make except a single call can only reference a single operation which will enable us to make an operation call and include parameters.

Another benefit with this approach is that you could alias the path to runner.py and then run any command from anywhere in NTRT. For example, right now if you want to build you need to cd into bin/build and then run build.sh. If you alias runner.py to, say, 'run' you would be able to do any operation in any directory. E.g. Calling 'run build' from anywhere within NTRT would cause a build to occur.

vsunspiral, 2015-02-13:
I like it.  I think though that instead of calling it "runner.py" we could call it "NTRT.py"

vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

PerryBhandal, 2015-02-13:
NTRT.py sounds good!

kcaluwae, 2015-02-13:
It's not really my call, but did you guys look into python setuptools?
https://pypi.python.org/pypi/setuptools

It's the standard way to install things in python and it's also compatible
with pip etc.

The syntax is pretty much exactly what Perry proposed:

python setup.py install

PerryBhandal, 2015-02-13:
My impression was that setuptools is supposed to be used for installing, uninstalling, etc, a Python package. In this case we want the ability to add any arbitrary NTRT operation to it (e.g. we want to be able to call ntrt flint to have flint run on the entire repo, or ntrt flint -r . to run flint in the pwd and for all sub folders. 

Am I mistaken in my assumption that setuptools isn't suitable for that use case?

ryan1248, 2015-02-14:
That sounds good to me!

An alternative would be to have it just be 'ntrt' (no extension). If you
put '#!/env/bin python' at the top and set it as executable you can just
call it directly.

Then you could do things like "#> ntrt setup --verbose" or "#> ntrt build
/src/dev/radams"

Thoughts?

kcaluwae, 2015-02-14:
I haven't tried it myself, but customizing setuptools doesn't appear too
difficult.
See for example:
http://ziade.org/2007/09/30/extending-setuptools-adding-a-new-command/

That being said, pick the tool (or custom code) that you think works best
(I haven't used NTRT myself in a while).
I just wanted to make sure that you had considered setuptools, when I read
the examples in your email ("python runner.py setup").

Have a great weekend,

Ken

PerryBhandal, 2015-02-14:
Ryan, my only concern with the header shebang approach is we wouldn't be able to use it with NTRT's virtualenv install. Is there any way to do that? (Come to think of it, how can we ensure the virtualenv is activated if they simply call python with an absolute path to the NTRT.py script? I'm not sure if that's possible with my proposed approach.)

vsunspiral, 2015-02-14:
Even better!

--------------------------------------------
Vytas SunSpiral    
PI - Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to arrive safely at death.

vsunspiral, 2015-02-14:
I'm a big fan of using existing standard tools, rather than home brewing our own solutions. This looks promising. 

--------------------------------------------
Vytas SunSpiral    
PI - Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to arrive safely at death.

PerryBhandal, 2015-02-14:
My only concern with setuptools is that I want command autocompletion for all project operations. It looks like this Python library can help with that:

https://pypi.python.org/pypi/argcomplete/0.8.4

What I'll do is try to modify setuptools to include argcomplete and see if that can satisfy our needs. If it can't, we can discuss a custom solution. It'll be great if setuptools does the job though, that way we won't have to bother with processing command line arguments, creating our own --help arg, etc.

ryan1248, 2015-02-14:
Glad you're thinking in the virtualenv direction! Fortunately the shebang
will work with virtualenv (if you use the correct shebang, which I didn't
above...  :-)

You can test with this:
-- test_shebang.py ---
#!/usr/bin/env python
import sys
print sys.executable
---

chmod +x test_shebang.py

./test_shebang.py  # Should output the system python executable
source path/to/NTRTsim/env/venv/bin/activate  # activate your virtualenv
./test_shebang.py  # Should output the venv python executable

The way it works:
- Activating virtualenv changes some environment variables such that the
'python' command is remapped to your virtualenv's python
- The shebang calls '/usr/bin/env', then calls python. /usr/bin/env pulls
up the latest environment variables, then calls whatever python executable
is defined in your environment variables
- If you have not activated a virtualenv, the environment variable will
point to system python. If you have activated it, the environment variable
will point to your virtualenv python.

Hope this helps,
Ryan

PerryBhandal, 2015-02-14:
That is awesome! That will make this quite easy -- I'll get it finished and committed shortly.

Thanks Ryan.

ryan1248, 2015-02-14:
Another thought -- it might be a good idea to have shell scripts for
different operating systems that first creates the virtualenv, switches to
it, and runs the rest of the setup.

For instance:
-- setup.sh ---
#!/bin/bash

# Note: the following two commands will use the system python
python [script to set up the env directory].py
python [script to set up the virtualenv under env].py  # should download
and install the virtualenv

# activate the virtualenv
source env/venv/bin/activate

# Now we're in our virtualenv
python [script that handles the rest of the setup]
-----

-- setup.bat ---
@ECHO OFF
CALL python [script to set up the env directory].py
CALL python [script to set up the virtualenv under env].py

# activate the virtualenv
CALL env\venv\Scripts\activate

CALL python [script that handles the rest of the setup].py
-----

What do you think?

-Ryan

PerryBhandal, 2015-02-14:
For the other scripts we use which use addons from virtualenv, how will we go about activating it? Should we just use the virtualenv shebang you posted earlier? or

PerryBhandal, 2015-02-14:
Oh, wait, even with the header shebang we still need to activate it. Hmm.

I'm actually not sure how best to resolve that. We could have some script called init_venv.sh which initializes/activates the venv on the first run, then activates it on subsequent runs. Then when people alias the NTRT.py script they can put a call to init_venv.sh prior to it. There must be a cleaner way to accomplish this though.

PerryBhandal, 2015-02-14:
One possible approach is to have ntrt.sh and ntrt.bat in the main dir. When it's called, it first activates the virtualenv if necessary, then it passes its args onto ntrt.py.

ryan1248, 2015-02-14:
I think there a few options for ntrt.py:
1) The approach you outlined (having shell scripts that run ntrt.py)
2) Having ntrt.py check to see if it is running under a virtualenv (import
sys; if hasattr(sys, 'real_prefix') then we're in a virtualenv) and if not
then
   a) exit and display a message like "virtualenv must be activated before
running ntrt -- use 'source env/venv/bin/activate' and then run ntrt again"
   b) spawn a subprocess that runs ntrt.py in the virtualenv
3) call the virtualenv's python directly from the NTRT scripts when
invoking other scripts (e.g. instead of 'python setup_bullet.py' we would
call 'env/venv/bin/python setup_bullet.py')

I'm leaning toward option #2a or option #1. #2b seems like it could get
messy. #3 could work but we'd need to remember to use it in all of the
scripts. Thoughts?

Some code notes:
-- ensure_virtualenv.py ---
#!/usr/bin/env python
if not hasattr(sys, 'real_prefix'):
    print "You must activate the virtualenv before running this script ..."
    exit()
-----

-Ryan

PerryBhandal, 2015-02-14:
I like option one myself. It seems to be the most simple.