NASA-Tensegrity-Robotics-Toolkit/NTRTsim #135: JsonCPP error during build

PerryBhandal, 2015-02-15:
Steven posted details about an install failure to the dev list on 2/15/15, e-mail is titled "Error during setup.sh" and includes a screenshot of the error. My response

--
It looks like jsoncpp is being flagged as failing due to a test failure. When the test fails it returns a non-zero exit code to the make call which causes our setup script to terminate.

One thing to try is modify line 133 in setup_jsoncpp.sh (https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/master/bin/setup/setup_jsoncpp.sh#L133) and remove the or, so line 133 will just contain make.

Depending on how jsoncpp is handling errors and when the errors are happening (I assume given that tests are running that the build itself of jsoncpp is done, but the question is whether it tries to clean up its recent build or not) NTRTsim might still work, or it might not. I'm not sure how widespread jsoncpp's use is at this point within NTRT.

It's odd that there's a test failure there though -- clearly the values are equal. I'll take a closer look into it tomorrow and see if I can get jsoncpp's developer(s) to address it, or, if they're absent/non-responsive, we can add a patch in to skip that test.
--

We need to either have this addressed by the jsoncpp devs, or put a patch in place on our end.

PerryBhandal, 2015-02-15:
It appears to be a known (and since resolved) problem:

https://github.com/open-source-parsers/jsoncpp/issues/90

We're currently frozen on a prior version, so we don't have the fix.

I've created a new branch called jsoncpp_head which points to a version of JsonCPP that has the fix. Steven, could you have that user attempt an install from scratch and let me know the outcome?

Perry

PerryBhandal, 2015-02-16:
@brtietz @vsunspiral @ryan1248 

Do any of you have any suggestions/know of best practices for handling dependency versions? Should we be freezing on a specific stable release, or pointing to the most recent stable release for a given dependency?

Assuming we should freeze, when would we bother to move to a new version? Only when issues like this arise or we find some new feature we need in the new version? or is there some other criteria?

vsunspiral, 2015-02-17:
Yes, we should point to specific releases and not float on "most stable recent release"
we don't want to be surprised by something changing and then breaking the build.

And yes, we update dependancies as needed -- for example to deal with an issue like this.
Or, if we end up wanting to access new features, etc.

updating dependencies often takes some management, and sometimes causes other dependencies to be updated, etc.  

zoooom
vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

PerryBhandal, 2015-02-21:
Whoops, didn't see Vytas' response until now. That sounds good -- thanks Vytas.

Steven, I'm going to go ahead and close this issue as I'm assuming the above resolved the issue. I'll also merge the change into master.

brtietz, 2015-04-15:
This issue appears to have returned, at least for some build configurations (a Debian server Perry's letting me use): https://gist.github.com/brtietz/14c195e8efab57842266

Current work around is to just re-run setup.sh after the error, it appears JSON works for our purposes in spite of the error.

Other troubleshooting steps include changing json.conf to direct to JSON's master download link, and the 0.y version: https://github.com/open-source-parsers/jsoncpp/archive/0.y.z.zip