NASA-Tensegrity-Robotics-Toolkit/NTRTsim #137: setup.sh error: CMakeLists involving Neural Network

eajung, 2015-02-20:
When running setup.sh I get this error. I'm not quite sure if anyone else had this problem, but I'm confused on what I need to change, whether the directory for NeuralNet is being called correctly or not. Please help if you guys can!

[image]
[image]

PerryBhandal, 2015-02-20:
What script are you calling to install NeuralNet? Is it setup_neural.sh directly, or indirectly via setup.sh?

eajung, 2015-02-20:
Just setup.sh

PerryBhandal, 2015-02-20:
That's odd. Try removing the env directory, then re-run setup while piping the output to log.txt (./setup.sh > log.txt and send me that log.

eajung, 2015-02-20:
Sorry for the late response. I had to eat some lunch, and classes.

== Setup ============
- Starting configuration

Initializing Env directory...

Initializing CMake...
- CMake is installed (cmake version 2.8.12.2). Creating link under env/bin.

Initializing GMockTest...
- GMockTest is installed under prefix /home/erikjung/NTRTsim/env -- skipping.

Initializing JsonCPP...
- JsonCPP is installed under prefix /home/erikjung/NTRTsim/env -- skipping.

Initializing Neural Net...
- NeuralNet is already built under /home/erikjung/NTRTsim/env/build/neuralNet/nnImplementationV2/Neural Network v2 -- skipping.
- Installing NeuralNet under /home/erikjung/NTRTsim/env
[100%] Built target neuralNetwork
Install the project...
-- Install configuration: ""
-- Up-to-date: /home/erikjung/NTRTsim/env/lib/libneuralNetwork.a
-- Installing: /home/erikjung/NTRTsim/env/include/neuralNet/Neural Network v2
-- Up-to-date: /home/erikjung/NTRTsim/env/include/neuralNet/Neural Network v2/dataEntry.h
-- Up-to-date: /home/erikjung/NTRTsim/env/include/neuralNet/Neural Network v2/neuralNetworkTrainer.h
-- Up-to-date: /home/erikjung/NTRTsim/env/include/neuralNet/Neural Network v2/dataReader.h
-- Up-to-date: /home/erikjung/NTRTsim/env/include/neuralNet/Neural Network v2/neuralNetwork.h

Initializing Bullet Physics Library...
- Bullet Physics is installed under prefix /home/erikjung/NTRTsim/env -- skipping.

Initializing Boost...
- Boost is installed under prefix /home/erikjung/NTRTsim/env -- skipping.

Setup Complete!

* Next Steps: Use bin/build.sh to build your code. The results will be placed under the 'build' directory.

eajung, 2015-02-20:
Okay, your solution of removing the env folder then running the setup worked perfectly. The NTRTsim is working like a charm. Thank you!

PerryBhandal, 2015-02-20:
My plesure!

closed 2015-02-20 by PerryBhandal