NASA-Tensegrity-Robotics-Toolkit/NTRTsim #139: Question: Could you provide clear OSX install with requirements

irthos, 2015-03-15:
Hi, I'm having trouble installing on OSX Yosemite. Could you please clarify deps and required steps for installation.

setup.sh fails with the following error..
`Initializing Neural Net...
- NeuralNet package already exists ('/Users/i/Downloads/NTRTsim-1.1/env/downloads/nnImplementationV2.zip') -- skipping download.
helper_functions.sh: line 99: [: too many arguments
Unpacking neuralnet to /Users/i/Downloads/NTRTsim-1.1/env/build/neuralNet/nnImplementationV2/Neural Network v2 (this may take a minute...)
Archive:  /Users/i/Downloads/NTRTsim-1.1/env/downloads/nnImplementationV2.zip
replace nnImplementationV2/license.txt? [y]es, [n]o, [A]ll, [N]one, [r]ename: A
  inflating: nnImplementationV2/license.txt  
  inflating: nnImplementationV2/Neural Network v2.ncb  
  inflating: nnImplementationV2/Neural Network v2.sln  
  inflating: nnImplementationV2/Neural Network v2/dataEntry.h  
  inflating: nnImplementationV2/Neural Network v2/dataReader.cpp  
  inflating: nnImplementationV2/Neural Network v2/dataReader.h  
  inflating: nnImplementationV2/Neural Network v2/letter-recognition-2.csv  
  inflating: nnImplementationV2/Neural Network v2/main.cpp  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj.Bobby-PC.Bobby.user  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj.Coldon.Bobby.user  
  inflating: nnImplementationV2/Neural Network v2/neuralNetwork.cpp  
  inflating: nnImplementationV2/Neural Network v2/neuralNetwork.h  
  inflating: nnImplementationV2/Neural Network v2/neuralNetworkTrainer.cpp  
  inflating: nnImplementationV2/Neural Network v2/neuralNetworkTrainer.h  
missing header for unified diff at line 4 of patch
patching file CMakeLists.txt
missing header for unified diff at line 27 of patch
patching file dataReader.cpp
missing header for unified diff at line 40 of patch
patching file neuralNetwork.cpp
missing header for unified diff at line 224 of patch
patching file neuralNetwork.h
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- "NTRTmaster/env/build/neuralNet/nnImplementationV2/Neural Network v2/neuralNetwork.cpp"	2014-10-03 15:14:25.618566584 -0400
|+++ "NTRTTests/env/build/neuralNet/nnImplementationV2/Neural Network v2/neuralNetwork.cpp"	2015-01-07 13:12:33.331713575 -0500
--------------------------
File to patch: NTRTmaster/env/build/neuralNet/nnImplementationV2/Neural Network v2/neuralNetwork.cpp
NTRTmaster/env/build/neuralNet/nnImplementationV2/Neural: No such file or directory
Skip this patch? [y] n
File to patch: 
`

PerryBhandal, 2015-03-15:
Have you taken a look at the INSTALL file? It lays out the dependencies and the installation process. Or are you asking for more detailed instructions than what is listed there?

irthos, 2015-03-16:
I'm asking for more detailed instructions specifically for MAC, please, as the above error was the result of following INSTALL instructions, and running setup.sh. Thanks.

PerryBhandal, 2015-03-16:
Without seeing a log of the original installation attempt (i.e. the first call to setup.sh) it's tough to know what it is. Please do the following:

- Delete your env directory, and all files ending with .conf in the conf directory (you'll see .conf.default files in conf/default - be  sure to not delete those, only the ones with a .conf suffix and nothing more.)

- Run setup.sh and pipe its output to a text file: ./setup.sh > setup_log.txt

Once you post that log I can help you troubleshoot further.

irthos, 2015-03-16:
Thanks Perry, followed your instructions. Ran setup.sh once and it ended after creating new .conf files. Ran setup.sh again, piped, and here's the log output: http://pastebin.com/c33rDP7Y

PerryBhandal, 2015-03-16:
Thanks irthos. It seems to be related to the patches we have for neuralnet, but given that I don't have a Mac it's tough for me to troubleshoot further. @brtietz sent an e-mail to the dev list an hour ago with the hopes of finding a Mac dev who can help troubleshoot.

PerryBhandal, 2015-03-19:
Hi Irthos,
I'm sorry that no one has gotten back to you yet. Unfortunately it's not clear yet why the error is happening. While I'd like to be able to dig deeper, I don't have a Mac.

In the meantime one thing you can try is disabling the neural net install. You can do that by commenting out line 141 in setup.sh (https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/master/setup.sh#L141) to disable the neural net call during set up. Frankly, it's tough to say what will happen if you disable NeuralNet's install. It'll largely depend on what you end up doing with the simulator. If you want to do any learning work (especially anything related to @brtietz's work) it will likely encounter problems. For any other uses it... might work?

Your issue has made it clear that we have some deficiencies in how we handle OS X. Currently we have a continuous build system which would quickly discover any errors on Linux systems, but that isn't the case for OS X. That's not something we can address immediately, but we do hope to have nightly builds (at the least) being run on OS X soon so we can track down issues more quickly. @vsunspiral was kind enough to run a NTRT build on his OS X system, and he had a completely different build error that we're not seeing on Linux, so evidently our OS X problems are not limited to you.

I know several others devs are quite busy right now. I've been planning on completing our move to new Python based install scripts, but I likely won't have time to complete that until mid-April. It's tough to say if that will even resolve your problems! In the meantime, if you would like, I can provide you with my Google hangouts address (or Skype username) if you would like some help troubleshooting. I'm sure I could figure it out if I had a mac system to troubleshoot with live. If you're not in urgent need of NTRT for your work (or whatever your motivations!) I can alternately send you an e-mail once we have this issue resolved.

brtietz, 2015-03-19:
Irthos,

If you take Perry's suggestion, you'll need to comment out the
NeuroEvolution libraries and things that depend upon it in CMake. If you
comment out the folder src/learning/NeuroEvolution, the NeuroAdapter in
src/learning/Adapters and dev/brtietz, the rest of NTRT should build.

Brian

vsunspiral, 2015-03-19:
Another option would be to install a linux VM on your mac and run in there.  Several folks use NTRT in VM's, and it works fine…..   
that would probably be the easiest way forward.

Perry -- could you update out github documentation to indicate that OSX support is currently broken?

thanks
vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

PerryBhandal, 2015-03-19:
Oh, yes, @vsunspiral's suggestion is better than mine.

To make this easier, I'll pre-assemble a Linux image with NTRT installed and ready to go and host it on ntrt.perryb.ca along with instructions for installing VirtualBox, loading the image in, etc.

Vytas, I'll take care of updating the documentation now.

irthos, 2015-03-24:
Thanks guys. I'll install it on linux.

I did comment out neuralnet which allowed setup.sh to complete but then following bin/setup.sh after threw an error and exited. here's the log if it helps http://pastebin.com/WxksZ0GH

closed 2015-03-24 by irthos