NASA-Tensegrity-Robotics-Toolkit/NTRTsim #142: Support for non C++ and third party learning libraries (part 1: generic parameter input/output)

brtietz, 2015-03-19:
As we move towards multi process learning (#133), its becoming apparent that

1. The learning process needs to control the simulation, as opposed to the current situation where the controller queries the learning.
2. Other languages may make multi process and multi system distribution easier (such as dispy)
3. Adding interfaces for other languages will expand our user base

Therefore I'd like to write or incorperate a library that
* Recieves a config file as an input on the command line
* Reads from input (parameter) files according to that config file
* Provides NTRT's controllers (or structures) with parameters based on the input files
* Writes a(n) output file(s) based on that config file and NTRT's results

This should interface well with existing learning controllers, as they expect vector<vector<double> > or a Boost multiArray.

I think JSON is a good interface for the config file, but I am torn on whether its a good choice for the parameter files. Ideally we're just specifying an n x m matrix of parameters (or several), so this can be a lot more generic, like our existing .nnw files. As a counter-argument, it would be nice to provide the ability to hand tune learned parameters, for which JSON is a good fit.

Thoughts on a generic file format? Obviously it is possible to support multiple in the long run, I'm trying to get a good sense of everyone's requirements and desires.

Any recommendations on existing libraries or ways to adapt JSON?

brtietz, 2015-03-19:
Upon further investigation, it looks like JSON arrays will work pretty well: http://json.org/

Interfacing with the Neural Networks may take some custom code, but nothing too difficult.

vsunspiral, 2015-03-19:
I'm pretty sure that JSON will do the trick.  They key is that we are using named variables so that the config file can be very flexible and we can add whatever unique data is needed for some unknown future experiment, and that the order of data within the config file is not crucial so that we don't end up with fragile formats.

zooom
vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

brtietz, 2015-03-25:
I forgot to tag this issue in the pull request, but #144 will close this issue once merged.

brtietz, 2015-04-01:
Closed by #144

closed 2015-04-01 by brtietz