NASA-Tensegrity-Robotics-Toolkit/NTRTsim #143: Seeking a Mac dev to maintain NTRT's mac builds.

PerryBhandal, 2015-03-19:
We're currently seeking a Mac dev who can help maintain NTRT on OS X systems. NTRTSim is currently broken for OS X, so that's the immediate concern. Once we have it working, we would like someone who can help ensure it remains that way!

PerryBhandal, 2017-09-08:
Yep!

There is, admittedly, less activity/support around the project nowadays. That might make this a bit more challenging, but anyone who wants to give it a shot is welcome to do so.

jerryderry, 2017-09-12:
NTRT can now be built on macOS 10.12. I have forked the repository, but I changed a lot to make it work, and I didn't write any code checking platforms. So I'm not sure if I should pull request, as it may not be built on Linux.

PerryBhandal, 2017-09-12:
That's great to hear. Thanks for investing the effort.

I'm concerned about pulling the code into the main repo without some certainty that it isn't broken for existing users. We have a fairly significant restructuring coming with v1.2 (main difference is that NTRT will act more as a library/framework, and applications will be kept in separate repos) so we might be able to bring that in then.

I've started a build of your code on Linux (Mint 18.2) and will report back on whether it was successful, then we can go from there.

jerryderry, 2017-09-12:
I'm glad it will change into a framework. I actually hope that NTRT could be a proper application, where you build your models, visualize them, do simulations, check internal forces, etc, all in this application, without writing separate c++ files. And not just be used for robotics. I guess more like a universal platform.

PerryBhandal, 2017-09-12:
The build ended up failing on the NeuralNet setup step. The output can be found here:

https://gist.githubusercontent.com/PerryBhandal/18028846214b96020586862130f35652/raw/5a8d6c5741d2f661a8f05427893f5c360e2ff8c4/gistfile1.txt

I'd say our best path forward is to try and return OS X support with version 1.2. I can keep you in the loop on when that is happening/once the migration plan is more firm and from there we can sort out how to integrate your changes in a stable manner.

It will likely be some time before that migration happens, unless one of the other devs is able to handle it.. My plan is to continue working on the v1.2 transition after November 18th with an aim of having the migration complete before the end of the year. It's possible that some other dev might take on the work of finishing the migration so it might be doable sooner, but that doesn't seem likely given how busy everyone is at the moment.

I'll get in touch once we're closer to the migration and we can go from there. It'll be great if we can have someone managing the OS X side, so hopefully you'll have the time/interest then.

jerryderry, 2017-09-12:
I see what's going on with NeuralNet. In the original patches, there are plenty of codes that use `tr1` to generate random numbers. And this is before c++11 standard. Most of the functions are now in c++11. But on mac, the standard library doesn't ship with `tr1`, so clang can't recognize it, and the files have to be changed to conform c++11. That's why in the error message, it's saying you need to use `-std=c++11` to flag the compiler.

Perhaps you may want to change the function `build_neuralnet()` in the file `setup_neuralnet.sh`, and modify to `-DCMAKE_CXX_FLAGS="-std=c++11" \`?

PerryBhandal, 2017-09-12:
I can give that a shot shortly, but even if it can be made to compile it's probably wise to wait until 1.2 to return OS X support. One of the big problems with NTRT so far (and something we'll hopefully do a better job of in 1.2 and beyond) has been a lack of focused testing/QA. Often modifications are just merged into the master after a brief glance which has made the process of maintenance/ensuring stability more challenging.

jerryderry, 2017-09-12:
Yes that's fine by me. At least I have a working one for myself.

I agree with testing. Another issue is that I think more tutorials are needed. I've tried NTRT 3 years ago, but I still don't know how to use it. For example, I have no idea why the structure keeps vibrating even without controllers.

PerryBhandal, 2017-09-12:
No doubt that better tutorials are needed. The documentation in general could be far better, and hopefully make NTRT more accessible.

I believe there was a thread about the vibrating issue on the mailing list. I'll take a look once I'm home and if I can find the specific thread I'll forward you a copy of it.

apsabelhaus, 2017-09-12:
Hi Perry and jerryderry -

Happy to hear that Mac OS support is being considered again!

About the tutorials and structures vibrating:
1) Take a look at the video tutorial I put together last year. It's linked
in the README at the root of the NTRTsim directory. I think I discuss some
scaling there.
2) Here is the archive for the NTRT users list (
https://lists.nasa.gov/pipermail/ntrtusers/) and here's the one for NTRT
dev (https://lists.nasa.gov/pipermail/ntrtdev/). I'd look through the NTRT
users archive for 'vibration.' In short: the scaling is off on the system,
and it's over-tensioned. When the forces applied are too high, the physics
simulator doesn't converge. Change the pretension parameter in the App or
YAML file(s) for your simulation.

Thanks,
Drew

-- 
Drew (Andrew P.) Sabelhaus
Graduate Student Researcher | UC Berkeley | Mechanical Engineering
Berkeley Emergent Space Tensegrities (BEST) Lab
NASA Ames Research Center | Intelligent Robotics Group | Dynamic Tensegrity
Robotics Lab

My pronouns are he, him, his. (What's
<https://www.ccsu.edu/lgbt/files/PreferredGenderPronounsForFaculty.pdf>
this? <http://lgbt.williams.edu/resources/trans-resources/pronouns/> Why?
<https://www.aclunc.org/article/frequently-asked-questions-whats-pronoun>)
Phone: (301) 807-9842 | www.apsabelhaus.com
------------------------------------------------------------------------------------------

jerryderry, 2017-09-13:
Thanks guys, I'll have a scan of the archive!