NASA-Tensegrity-Robotics-Toolkit/NTRTsim #145: Self-host all dependencies

PerryBhandal, 2015-03-30:
Brian e-mailed me this morning about a user experiencing a compile issue under Linux related to NeuralNet. I've included the log he sent at the bottom of this issue.

The easiest way to address this might be to host at least some of our own dependencies, if not all of them. This is less important with Bullet or Boost where we can have a fair degree of confidence that they're not going to place new versions at the same URL. For packages like NeuralNet, it seems like we should find a version that works, add the version to its name on our own (even if it's just some internal revision number) and then self-host it.

I can offer up storage.perryb.ca for this. It already hosts the NTRT appliance download link, I include the link below in case anyone wants to check the speed on it.

http://storage.perryb.ca/ntrt/

At the least we should move NeuralNet (and any other packages that might have their versions silently modified) to that box. We can, of course, move things like Bullet and Boost onto it, but that's not really necessary.

If this change sounds good, let me know and I can create a user that has write privileges to that domain's FTP.

---
Output

Initializing Neural Net...
Downloading nnImplementationV2.zip to /home/nick/projects/NTRTsim/env/downloads/nnImplementationV2.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   406    0   406    0     0   5943      0 --:--:-- --:--:-- --:--:-- 15615
100   408  100   408    0     0   3293      0 --:--:-- --:--:-- --:--:--  3293
100  594k  100  594k    0     0   586k      0  0:00:01  0:00:01 --:--:--  586k
helper_functions.sh: line 99: [: too many arguments
Unpacking neuralnet to /home/nick/projects/NTRTsim/env/build/neuralNet/nnImplementationV2/Neural Network v2 (this may take a minute...)
Archive:  /home/nick/projects/NTRTsim/env/downloads/nnImplementationV2.zip
  inflating: nnImplementationV2/license.txt  
   creating: nnImplementationV2/Neural Network v2/
  inflating: nnImplementationV2/Neural Network v2.ncb  
  inflating: nnImplementationV2/Neural Network v2.sln  
  inflating: nnImplementationV2/Neural Network v2/dataEntry.h  
  inflating: nnImplementationV2/Neural Network v2/dataReader.cpp  
  inflating: nnImplementationV2/Neural Network v2/dataReader.h  
  inflating: nnImplementationV2/Neural Network v2/letter-recognition-2.csv  
  inflating: nnImplementationV2/Neural Network v2/main.cpp  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj.Bobby-PC.Bobby.user  
  inflating: nnImplementationV2/Neural Network v2/Neural Network v2.vcproj.Coldon.Bobby.user  
  inflating: nnImplementationV2/Neural Network v2/neuralNetwork.cpp  
  inflating: nnImplementationV2/Neural Network v2/neuralNetwork.h  
  inflating: nnImplementationV2/Neural Network v2/neuralNetworkTrainer.cpp  
  inflating: nnImplementationV2/Neural Network v2/neuralNetworkTrainer.h  
patching file CMakeLists.txt
patching file dataReader.cpp
patching file neuralNetwork.cpp
patching file neuralNetwork.h
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- "NTRTmaster/env/build/neuralNet/nnImplementationV2/Neural Network v2/neuralNetwork.cpp"	2014-10-03 15:14:25.618566584 -0400
|+++ "NTRTTests/env/build/neuralNet/nnImplementationV2/Neural Network v2/neuralNetwork.cpp"	2015-01-07 13:12:33.331713575 -0500
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
2 out of 2 hunks ignored
- ERROR: Failed to patch NeuralNet, 2nd patch
Neural Net initialization failed -- exiting now.

brtietz, 2015-03-30:
Regardless of whether we adopt this solution, we should include completely removing env as a troubleshooting step in INSTALL.

There may also be ways to make the patching procedure more robust.

vsunspiral, 2015-03-30:
I agree with this -- we need to have stable packages.
if you can host it Perry, that would be ideal.
for various reasons it would be best if we were not directly redistributing 3rd party packages directly as part of our repository….

vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

PerryBhandal, 2015-03-30:
Great, okay -- yep, I can host it!

I'll merge PR 146 which contains the new URLs.

closed 2015-03-30 by PerryBhandal