NASA-Tensegrity-Robotics-Toolkit/NTRTsim #154: Feature Request: Video Walkthrough for Install and Basic Use

irthos, 2015-04-23:
Feeling a bit overwhelmed with the install and use of this sim but really excited to get into it. If it isn't that much trouble, it would be great to see a video of how to get up and running, along with some really basic examples of how to use this to build tensegrity structures. If a vid like this already exists, please link it. Thank you very much!

PerryBhandal, 2015-04-23:
Hi Irthos,
Improving the resources for beginners is something we really want to do, there just hasn't been anyone that has had the time to spend a lot of time on it yet. That is definitely something we want to change, though.

For basic examples, your best bet is to look at 3_prism in the examples directory. That'll give you a good sample of how to instantiate a structure.

It's possible that @apsabelhaus may be more helpful than me. I know he helps users get started with NTRT at UC Berkeley. I'm not sure if he does that by helping people in person or if he might happen to have some videos, but hopefully he can chime in either way.

I'll be sure to let you know once I've made some progress on the tutorials I have planned. I have a few other improvements I'm hoping to finish first, so it likely won't be something that is ready soon, but I'm confident it can be available by late June.

In the meantime you can post questions to the users list if you would like as you get started, or you can message me on Google hangouts if some real-time assistance would be helpful.

irthos, 2015-04-23:
Thank you lots, Perry. perhaps we can do a hangout to walkthrough the steps and record the screencast, when you're ready, I'm willing to take on the work to edit the screencast into a concise tutorial (fast-forwarding the setup.sh script and removal of my dumber questions :) ) and then publish it on NASA or NTRT youtube channel, after your approval.

Also thanks for the suggestion of the user list and offer to assist in real-time! I'll keep tinkering for now and please feel free to connect with me on g-chat if you'd like to collaborate to make good tut videos (irth03 at gmail).

PerryBhandal, 2015-04-23:
That's a great idea -- definitely happy to do that!

There isn't currently an NTRT YouTube channel and that's not something I'd have the authority to create, but hopefully @vsunspiral or @brtietz can chime in. I think it would be a great place to store tutorial videos.

My immediate priority is getting the new setup and build scripts done as that will change the setup process a bit. I'm hoping to have that done by early next week (that'll also resolve some of the niggling OS X problems you encountered) and then we can arrange a hangout.

I'll also be able to run you through some examples at the same time.

irthos, 2015-04-23:
Wonderful, Perry! i'm excited to participate. until then, good luck with the improved scripts/installer!