NASA-Tensegrity-Robotics-Toolkit/NTRTsim #155: NTRT Youtube Channel

brtietz, 2015-04-27:
I wanted to pull this out as a separate discussion from #154. Do we want a centralized NTRT youtube channel (or similar)? If so, what videos go there that don't go on individual researcher's channels?

PerryBhandal, 2015-04-28:
I was originally the one that suggested it, but I'm having second thoughts about it since. My thought initially was that it would mainly be used as a central place to store tutorial videos, and maybe some of the more polished NTRT videos like the one by NASA 360.

As I've explored doing tutorial videos, I've become less keen on it. It's fine when you have a big project if videos go out of date as new software is released as someone can find time eventually to update them (and there is often a community effort to do so), but having one of us going through and updating the videos when necessary seems like a poor use of our time.

So, in short: a centralized NTRT channel would be worthwhile if we had a lot of tutorial videos, but we have so many pressing issues that take priority over that right now. As a result I don't believe it's worthwhile, at least presently. Though we have some enthusiastic folks (like Irthos) who are happy to spend the time cleaning up and preparing a video, but even in a case like his we have so many pressing issues where he could be more helpful. One way we could go about it is I can give Irthos help over a hangout to get started, and I'd be thrilled if he has the time to put it together into a written guide -- if not, one of us can certainly take care of it. Once Irthos is done he could submit a PR and we could make any final modifications necessary with ease. With a video modifying his work would be quite difficult.

vsunspiral, 2015-05-21:
It certainly wouldn't hurt to have an official NTRT video channel.  
I would suggest that we use it primarily for our higher quality videos.  
Paper videos that involve lots of NTRT work.
Or other "AHA!" videos that highlight some new feature or breakthrough.
lots of the "current status of my experiments" videos should still go to our personal channels.
but if you find yourself using a video in presentations, or otherwise wanting to share it with others, then it is probably "significant" and could be posed on the official channel.

I suggest that we just start doing it and let the channel fill up as it does.  I'm ok with a low level of managerial oversight (i.e. folks should just post away).

zoooom
vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.