NASA-Tensegrity-Robotics-Toolkit/NTRTsim #159: Specific versions of Ubuntu Dependencies

srlessard, 2015-06-18:
Is there a specific version of the following packages required?
- freeglut3
- freeglut3-dev
- libglib2.0-dev
- build-essential

I only ask because I don't have sudo access on my machine and it is a pain to update each one (but currently, neither my setup.sh nor copied binaries from other repository clones properly function)

brtietz, 2015-06-18:
Steve,

Can you delete env and build, and post the output of setup.sh? Deleting env
and build will solve 90% of problems, and seeing the errors will help us
diagnose.

Brian

PerryBhandal, 2015-06-18:
(For posterity, here's a gist of Steve's setup failure: https://gist.github.com/PerryBhandal/03de26838871e6426e8e)

This is a separate issue from the float test failure we saw in the past.

PerryBhandal, 2015-06-18:
Steve, so I can understand as I refactor some of the setup: you mention that it's a pain to update each one, how exactly do you handle cases where you need to update a dependency? Are you needing to request permission to have it updated, or do you do some sort of local compile of the correct version then use that?

srlessard, 2015-06-18:
I have to request permission from NASA's IT department. It's not that big of a deal, I just wanted to avoid the delay in getting installations (they usually take a business day to install)

srlessard, 2015-06-18:
Linking CXX executable jsoncpp_test
./jsoncpp_test: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by ./jsoncpp_test)

Is the error message I get during setup

PerryBhandal, 2015-06-18:
Alright, thanks. While this won't help you immediately, I think we can do more to support users that don't have administrative permissions.

Unfortunately I don't know what's causing your issue. I'll let you know if I come up with anything.

srlessard, 2015-06-18:
Ok, thank you

PerryBhandal, 2015-06-18:
Can you run *dpkg-query -l* and post the results as a gist?

srlessard, 2015-06-18:
<script src="https://gist.github.com/srlessard/06a577dccfcb426208ef.js"></script>

For libstdc++6

PerryBhandal, 2015-06-18:
Its entire contents would be ideal. Not sure if there is an upper limit on gist size, if there is feel free to e-mail me it as a text.

One other thing you can try if you're just trying to get some work done is this NTRT build:

http://ntrt.perryb.ca/storage/builds/ntrt_6.18.15-steve_build.tar.gz

It's entirely possible it'll fail (it's just the version of NTRT from the build server, which is built under 64 bit Debian), but I figure it's there if you want to give it a shot.