NASA-Tensegrity-Robotics-Toolkit/NTRTsim #163: tgHillyTerrain is (still) compiler dependent

brtietz, 2015-08-04:
My integration test which observes how far a tetraspine robot travels on hills is giving different values on buildbot and my local machine. Tests on flat ground (spinesTest) are passing. 

My computer: Dist travelled 5.19875
Buildbot: Dist travelled 5.37786

If you update to the latest version of master, can you post your values here (bin/build.sh -g)? Versions of g++ are also appreciated.

PerryBhandal, 2015-08-04:
I have GCC 4.8.2 locally. Doing a build of the head now, and I'll let you know if I get conflicting values. If I do, I'll update locally to 4.8.4 and let you know if that resolves it.

PerryBhandal, 2015-08-04:
I appear to be getting the same values as you did (5.19875) on my system with GCC 4.8.2.

gcc -v: https://gist.githubusercontent.com/PerryBhandal/0e66351f5a2770a1e0a4/raw/7aeb7016d710d734de5fec9e86dc5a731e141d89/gistfile1.txt

Output from build.sh -g: https://gist.githubusercontent.com/PerryBhandal/23e0bbf50c7d3fc32d0c/raw/3638aff6f588d3ec20641987f1e17739623c0497/gistfile1.txt

uname -a: Linux perry-laptop 4.0.4-040004-generic #201505171336 SMP Sun May 17 17:37:22 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

PerryBhandal, 2015-08-04:
Here's the info from the BB box:

root@ntrt:/# gcc -v; uname -a
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.2-19ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) 
Linux ntrt 2.6.32-26-pve #1 SMP Mon Oct 14 08:22:20 CEST 2013 x86_64 x86_64 x86_64 GNU/Linux

brtietz, 2015-08-05:
A little more info after some digging: The old (sequential impulse) solver doesn't have this problem. I suspect upgrading to Bullet 2.83 might solve this issue.

PerryBhandal, 2015-09-05:
Could we disable the problematic test in the meantime? We'll eventually want to sort the issue out, but in the meantime with the BuildBot flagging our builds as failing, we might overlook a new problem that arises in subsequent commits.

brtietz, 2015-09-06:
Perry - seems like a good idea, especially with all of the changes we
discussed last Tuesday.

PerryBhandal, 2015-09-06:
HillsTest has been disabled, and BuildBot has returned to a passing state.

Thanks Brian.

closed 2015-09-06 by PerryBhandal