NASA-Tensegrity-Robotics-Toolkit/NTRTsim #166: Installing Boost using setup.sh

chiku00, 2015-09-05:
I get an 'Install failed' error while installing boost.
last 8 lines:
...
"       ^
gcc.compile.c++ bin.v2/libs/wave/build/gcc-4.8/release/link-static/threading-multi/cpplexer/re2clex/cpp_re.o
gcc.archive bin.v2/libs/wave/build/gcc-4.8/release/link-static/threading-multi/libboost_wave.a
common.copy /home/arnab/Dropbox/DDP/Simulation_software/NTRTsim-1.1/env/lib/libboost_wave.a
...failed updating 6 targets...
...skipped 18 targets...
...updated 11024 targets...
Install failed
Boost initialization failed -- exiting now.
"
This happens for both this version as well as version 1.1

PerryBhandal, 2015-09-05:
If you run setup.sh again and the setup completes, you should be good to go.

It's been a known issue for quite some time (see issue #2).

PerryBhandal, 2015-09-05:
Oh, looks like you closed the issue just as I responded. I assume you noticed the same behavior when you ran it again!

In any case, if you run into any further problems, please feel free to open this issue again.

closed 2015-09-05 by PerryBhandal

chiku00, 2015-09-06:
Indeed. Mins after I posted it, I realized my mistake. Turns out I couldn't delete this issue altogether. Thank you PerryBhandal