NASA-Tensegrity-Robotics-Toolkit/NTRTsim #167: Issues in building AppPrismModel

chiku00, 2015-09-05:
First of all, at least 6 header files along with their cpp files had to be moved to /../../core before I reached here:
[image]

I have no idea how to proceed from here on. It would be greatly appreciated if someone has a solution for this.

PerryBhandal, 2015-09-05:
To build the applications you'll want to go into bin and run build.sh instead of directly compiling the source files. You can then execute the specific application to launch that structure in the simulator.

chiku00, 2015-09-06:
Aha! Yes it did work. This had in fact occurred to me before, but that time, I had made a separate folder in src/examples with the codes and launched build.sh but nothing happened. I was under the impression that all the folders in the examples directory would be converted into an executable.

closed 2015-09-06 by chiku00