NASA-Tensegrity-Robotics-Toolkit/NTRTsim #168: Regarding Co-ordinate system

chiku00, 2015-09-07:
Not really an issue, but I just noticed that the x,y,z directions are a little messed up. I looked at the 'move' function to position the object at the initial stage and I found this:
Edge, width, Height = X (+ towards the left),Z (+ towards the top, away from the surface),Y (+ away from the screen) with respect to the 3D Plane
The X,Z,Y labeling is what I came up; X going from left to right (I expected), Z going from bottom to top (I got) and Y going away from the screen (Right hand rule). It's different from what I see in PrismModel.h; the comments on line 108, assuming that this same ordering (x,y,z as per the comments) is used everywhere.

Nothing major, but I guess I was used to the co-ordinate system I was taught in High school.

brtietz, 2015-09-07:
Our coordinate system matches that of Bullet Physics, with Y as up:
http://bulletphysics.org/mediawiki-1.5.8/index.php/Coordinate_system

This is the convention within btVector3, which is also reflected in the
addNodes function.

vsunspiral, 2015-09-07:
> but I guess I was used to the co-ordinate system I was
> > taught in High school.

It turns out that many diffident industry use different standards for how they use coordinate systems.  There is no "global" standard, and what you were taught in high school was one of many possible standard labelings of axis.  

vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

closed 2015-09-10 by brtietz