NASA-Tensegrity-Robotics-Toolkit/NTRTsim #169: Minimum value for density, rod radius in AppPrism?

chiku00, 2015-09-07:
I was trying to make the whole model stiffer such that if I toss it, it won't collapse. I decided to reduce the rod radius, since it was 0.31 m (way too large for my model) to around 0.031 m (pretty reasonable). Turns out, upon compilation, the model was not visible at all. It does it have something to do with double precision and all? Reducing the density by a factor 10 also had the same effect.
Here are my input parameters for a working model:
 0.2,     // density (mass / length^3)
       0.31,     // radius (length)
       1000.0,   // stiffness (mass / sec^2)
       10.0,     // damping (mass / sec)
       5000.0,     // pretension (mass * length / sec^2)
       10.0,     // triangle_length (length)
       10.0,     // triangle_height (length)
       10.0,     // prism_height (length)
Moreover, is the stiffness entered above that of the tension ed string or the rod? (it appeared to be that of the string) Then where do I set the stiffness of the rods? Or are they not compressible at all?

brtietz, 2015-09-07:
See this article on scaling:
http://www.bulletphysics.org/mediawiki-1.5.8/index.php?title=Scaling_The_World

Note that the default 3_prism is run at cm scale (Gravity 981 cm/sec^2).
This is changed at the app level. This is why there aren't specific units
in the comments.

The rods are assumed to be rigid (not compressible), therefore the the
stiffness only applies to the string.

Brian

brtietz, 2015-09-10:
Leaving this open in case we want to adopt these default values.

chiku00, 2015-09-17:
I tried to decrease the value of density to 0.1 and there the model disappears from the screen. It would then seem that it is in our interests to use grams for mass instead of kilograms.

vsunspiral, 2015-09-17:
See the discussion about Scaling:

http://ntrt.perryb.ca/doxygen/scaling.html

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.