NASA-Tensegrity-Robotics-Toolkit/NTRTsim #170: Upgrading to Bullet 2.83

brtietz, 2015-09-10:
I took two hours to attempt the upgrade to Bullet 2.83, and here's where things stand:

Our setup scripts do just fine, much of the physics and simulation API is the same.

The rendering API has totally changed, as has the examples API. Examples are now passed to an Example Browser, which reads in a list of examples and can run them all.

In my first attempt, I tried making tgSimViewGraphics a subclass of OpenGLExampleBrowser, and calling the appropriate functions. This included some questionable passes of NULL, and I'm not confident this is the right way to go.

The current roadblock to this approach is that OpenGLExampleBrowser is not packaged up in a library, so I get linker errors when I try to compile. I'm really hoping to avoid patching Bullet this time around, so we'll need to make our own library for this purpose. I still think OpenGLExampleBrowser has a lot of functions we need.

I haven't figured out what the current equivalent of btIDebugDraw is (for rendering cables and markers), so there's some more work to do there as well.

I think we may give up too much low level control by re-factoring tgSimulation in accordance with the Bullet examples. They do not appear to have a step call, and we need that. This may be solvable with lots of subclassing, which we may need to do anyway.

This branch has the current state, I may have more time for this next week:
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/Bullet283

brtietz, 2017-02-08:
Bullet has released 2.86: http://bulletphysics.org/wordpress/?p=472

vsunspiral, 2017-02-08:
that looks like it has some really interesting new features — specifically the use of pybullet:
https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3 <https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3>

Key things:
With pybullet you can load articulated bodies from URDF, SDF and other file formats. pybullet provides forward dynamics simulation, inverse dynamics computation, forward and inverse kinematics and collision detection and ray intersection queries.

<vytas> URDF is a really useful format that came out of ROS, and is the go-to format for describing traditional robots for simulation.  useful if we wanted to do comparisons with traditional legged robots, for example, and one should be able to quickly find existing URDF models of robots….

Another key thing:
pybullet wraps the new Bullet C-API <https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsClientC_API.h>, which is designed to be independent from the underlying physics engine and render engine, so we can easily migrate to newer versions of Bullet, or use a different physics engine or render engine. By default, pybullet uses the Bullet 2.x API on the CPU. We will expose Bullet 3.x running on GPU using OpenCL as well.

And finally:
pybullet can be easily used with TensorFlow and frameworks such as OpenAI Gym.

we built up our own neural network toolset before modern things like TensorFlow were available.  Going forward, it could be good to migrate to using TensorFlow, when we update to the current version of bullet.  When that will be is less clear, but I’m working on an NSF proposal and a NASA project proposal that would both explicitly get us funding to support NTRT.

vytas

> 
> Bullet has released 2.86: http://bulletphysics.org/wordpress/?p=472 <http://bulletphysics.org/wordpress/?p=472>

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.