NASA-Tensegrity-Robotics-Toolkit/NTRTsim #171: Rod structure collapses into a 2d

chiku00, 2015-09-16:
I tried to build a structure purely out of rods, similar to the tetrahedron model; but it always collapses into a 2d model. Here is how it looks like:
[image]

The Co-ordinates are in 3d. R=8 and h = 12.
Here's a link to the files; headers and all:
https://www.dropbox.com/sh/nvrygejqlxntjvg/AAAADBPFhzyV47XI34mGJyVWa?dl=0

chiku00, 2015-09-17:
I am trying to make the tensegrity model on the left below. The one on the right is the same model in Matlab.
[image]
When I try to build it in NTRT, it always ends up giving me this result:
[image]
In the same shared folder indicated in the above post, there's a folder 'Model' with the Matlab model and the NTRT code. The Matlab model was only to see if indeed the model was false; but appears to be fine. 
I tried to look at the model of the Tetrahedral snake in the examples. I don't see any difference in the instructions I gave and what is present here. Even after removing the tags ( a wild guess), it still gave me the same model (tetrahedron). This is, of course, when I set the number of elements in the snake to one.

vsunspiral, 2015-09-17:
hey,
So, my guess is that you need to work on the getting the right balance of tensions in the cables for the NTRT model to stand up.  I have not looked at your matlab code, but I suspect that you are defining positions for the rods and then then drawing strings between them, but do not take into account the forces, elasticity, and motion of the resulting structure.  In NTRT, once you create the structure it is then subject to physics and will collapse if there is not enough pre-tension to keep it up.  

there is a mechanism to set the pre-tension in the cables as you create the structure.  I would try tightening that up.

zoooom
vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

chiku00, 2015-09-17:
When I say collapses, the resultant structure does not consist of individual rods, but are attached to one another. 
I've shared the resultant executable in the folder. 
From the Matlab pic, you can see that the rods (red lines) are attached at one point. There are 6 in all for that one sub-structure. However, in the resultant structure, that substructure lies in one plane entirely. 
It would have been fine had it collapsed because of loading, but I can't figure out why this result came to be. BTW, the matlab file is only to generate how the structure should look; nothing to do with a physics simulation. I just wanted to check that the structure I wanted to create is indeed what those nodes, rods and threads would create; and not some mistake in taking nodal position values.

chiku00, 2015-09-18:
It appears that trigonometric functions are not defined while executing the code. It didn't give me any errors while compiling, so I had assumed that the failure of the model was due to something else.
[image]
Now I'm concerned about data acquasition #74 .

vsunspiral, 2015-09-18:
Could you file an issue about the trig functions so that we remember to investigate as time allows?  Thanks!

--------------------------------------------
Vytas SunSpiral    
PI - Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to arrive safely at death.

closed 2015-09-22 by chiku00