NASA-Tensegrity-Robotics-Toolkit/NTRTsim #173: Tags: only alphabetical

chiku00, 2015-09-20:
I tried to use numeric tags, as it allowed me to include information regarding nodal position when I needed to apply loading on that string.
Apparently, I can only use alphabets. But, correct me if I'm wrong, a char string can store numbers but in the form of characters; meaning they can't be used in numerical equations.
It was compiling fine, but the executable was not opening. Hence, the diagnostics.

ryan1248, 2015-09-21:
Hi,

The only requirement is that the tag is not *entirely* numeric, so "A1",
"A2", etc. will work just fine. I would recommend this approach for your
use case.

By the way, the reason that integer-only tags are not allowed is so that it
is possible to reference different aspects of a structure by both array
position and associated tags even if the info is coming from a string-based
representation (like JSON). It also reduces confusion and helps avoid
hard-to-spot bugs.

Hope this helps,
Ryan

chiku00, 2015-09-21:
Thanks. I replaced the numbers with their corresponding alphabet. eg: 1 ==a; 26 == z; 27 == aa

closed 2015-09-21 by chiku00