NASA-Tensegrity-Robotics-Toolkit/NTRTsim #174: Set Pretension, stiffness while using controller

chiku00, 2015-09-20:
Is it possible to set pretension and stiffness of the threads while using a controller? I'm using the impedance controller but I can't figure out where I should give these values in addition to damping. I had opened 'tgImpedenceController' and found variables like 'kDefaultOffsetTension' , 'kDefaultLengthStiffness' and 'kDefaultVelocityStiffness'. They possibly mean 'Pretension', 'Stiffness' and 'Damping', respectively, but upon putting in these values (50000,100000,10), I'm not getting the desired output in terms of the behavior of the system. The picture below describe what I mean: the one on the right is with the same set of values but without the controller; the other is with the controller.  
[image]

chiku00, 2015-09-20:
The code for both of them are here:
https://www.dropbox.com/sh/k65pijdiy1kgmr8/AAC1iEf6BycsyGXjQBtYneFea?dl=0
For the model on the left, files are in '3_prism'; the other on is in 'NestedTetrahedrons'.

brtietz, 2015-09-21:
The stiffness of the thread cannot be changed while the simulation is
running, that is a material property.

The pretension is changed by these controllers. In the case of the
impedance controller, this refers to the position gain, the velocity gain,
and the tension offset of the controller. These variables should probably
be renamed to avoid confusion. The equation for how these are summed into a
single tension gain is on line 174 of
controllers/tgImpedanceController.cpp.

If you just want to set the tension directly (Even change it over time) you
can just use tgTensionController, since it avoids including position
information,

chiku00, 2015-09-22:
What I meant was where can I set those material properties: damping and stiffness?

brtietz, 2015-09-22:
Those are part of tgBasicActuator::Config, line 163 of
NestedStructureTestModel.cpp
The entire config is viewable at core/tgSpringCableActuator and includes
motor parameters.

chiku00, 2015-09-22:
Apparently, the density was pretty high. I lowered it to 0.03 and I got back the previous level of openness, like the figure on the left in the top image. And, I'm able to notice the slight motion of the rods in response to the controller input.

closed 2015-09-22 by chiku00