NASA-Tensegrity-Robotics-Toolkit/NTRTsim #175: How to use Crater Deep and others?

chiku00, 2015-09-21:
Only 'rgHillyGround' has worked out for me by following the code in 'tgBoxGround'. But not so for CraterDeep and CraterShallow. If I want to make a custom obstacle ground, how do I go about doing that? Like in the pictures below:
[image]
[image]

srlessard, 2015-09-21:
Hi chiku00,

When you say that the crater terrains haven't been working out for you, do you mean they do not compile? Or they do not appear as intended? Regardless, the green and blue objects seen on the ground in those simulations are tgBoxes and tgSpheres (see src/core). You can mimic the pictured terrains by following the examples in Brian's learning controller programs (he is now cc'd).

Best,
Steve

brtietz, 2015-09-21:
The boxes and craters are referred to as obstacles, not terrains. They are
added similarly to models, such in the example obstacles app:

tgBlockField* myObstacle = new tgBlockField();

> // Add the model to the world

> simulation.addModel(myObstacle);

chiku00, 2015-09-22:
Upon going through the master file of NTRTsim, I see many new things have been added that were not there in the one available in version 1.1 (available through the NASA website), particularly folder 'Models' under 'src'. I was having trouble installing the master version, so I had decided to stick with the one available at NASA.

brtietz, 2015-09-22:
Do you remember what those errors were? If you could provide them, along
with your operating system info, that'd be very helpful.