NASA-Tensegrity-Robotics-Toolkit/NTRTsim #176: Log File Time

chiku00, 2015-09-26:
Just to be sure, the units under time in the log file is seconds, not microseconds, right? Because the last entry in the time column was 109587, and I know I didn't leave the simulator on for 30 hours. So what happened?

chiku00, 2015-09-26:
Also, can i not get the nodal co-ordinates?

brtietz, 2015-09-27:
It should be seconds. What are you passing in for dt when calling
tgDataObserver::onStep?

brtietz, 2015-09-27:
That's a bug - the center of mass of coordinates of each rigid body are
given, we'll have to enhance the logger in order to give nodal coordinates.

chiku00, 2015-09-28:
Well, 'dt' was 1/1000 when it was declared during 'AppNestedTetrahedrons'. In any case, those numbers just can't be in seconds; they have to be milliseconds.
https://www.dropbox.com/sh/ic6zo4gf5a6nqve/AACOffGi1cQPBDzjYLx_Ab1_a?dl=0
Here, under 'Data', you'll find the log file corresponding to the code in 'Code'. I've transferred the data  to Excel for plotting purposes.

brtietz, 2015-09-28:
This line is the problem

m_dataObserver.onStep(subject, simTime);

Pass dt instead of simTime and you'll get more reasonable results. The
functions almost always want time since last updated instead of current
time.

Brian

chiku00, 2015-10-03:
Well, that did the trick. 
Regarding the nodal positions, can you instruct me how to go about modifying the logging file to get the coordinates?

closed 2015-10-07 by brtietz