NASA-Tensegrity-Robotics-Toolkit/NTRTsim #177: Controller Input Parameters

chiku00, 2015-09-26:
Regarding the controller it self:
"
double setTension = _controller->control(*(stringList[i]),
                                            dt,
                                            Length,
                                            target
                                            );
" (NestedStructureSineWaves.cpp)
What does the parameters length and target mean in the model? I wanted to give it a unit ramp input by providing target the speed (there was a variable offsetspeed associated with target, hence my assumption). But the movements were negligible on the screen.
Then I set it to 12.5. Then at least i observed something on the screen. But I used an if condition to set it at simTime between 2 and 4 (seconds). But the data I pulled out, after importing it to excel, showed no changes in the lengths of the actuator (here, actuator 'a') in terms of 12.5 magnitude. Is the magnitude getting scaled down somewhere?
My overall objective right now is this:
Given I want to displace a particular node in a particular direction, which actuators should I actuate to achieve the target motion?
I need this to build a working prototype.
I'm trying to find a transfer function by studying the kinds of displacements that occur in the nodes for a unit actuation of an actuator.

vsunspiral, 2015-09-26:
> Given I want to displace a particular node in a particular direction, which actuators should I actuate to achieve the target motion?

This is a very hard problem and there is not an easy answer to it, other than making the general observation that if you want to precisely control the absolute position of a single node, you will likely need to control *all* the actuators to accomplish that exact trajectory.

This is when the research begins.

out of interest, what is your overall objective?  It would be fun to know what you are trying to accomplish!

vytas

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

chiku00, 2015-09-26:
My objective is to be able build a spider-bot, with tensegrity appendages. This will allow the bot to traverse various types of terrains, swim through water (octopus) and curl up into a ball for defense. 
[image]
The above image depicts the appendage. For as a bot, it could also help in repairing structures (static or moving) irrespective of where it is.
Now that I know that that itself is the current barrier in controlling tensegrity structures, I guess i need to overcome that before I get to building it.
I'm only competent with aspects related to mechanical engineering (Kinematics, dynamics, stress fields). I wonder how much can I contribute in the this field as it involves controls. But I do see a means to achieving it.