NASA-Tensegrity-Robotics-Toolkit/NTRTsim #178: Nodal Positions

chiku00, 2015-10-04:
There needs to be a provision for logging nodal positions. I traced a path all the way to btRigidBody but still no sign of a nodal value. Guessing they're being used by the forcing and torque functions else where before the net summation is sent over to each rigid body.

brtietz, 2015-10-07:
You could log each btRigidBody's orientation (via virtual btVector3 orientation() const; in tgBaseRigid, which is inherited by tgRod), and use that to compute the positions of the end points.

Alternatively, there is the abstract marker class in core. It may be a little tedious to implement that for every node position. We're also hoping to get rid of that class in the next few months and replace it with something better. You could help!

chiku00, 2015-10-07:
Of course. Any way I can. But as you can clearly see, I needed all the help you could give just to get where I am with the program.