NASA-Tensegrity-Robotics-Toolkit/NTRTsim #179: Doubt regarding the data in logging file

chiku00, 2015-10-05:
I tried to figure out why there was a decrease in tension when I increased the rest length. I thought that was due to the fact that stiffness = Youngs_Modulas*Area/Rest_Length. Since rest length increased, stiffness decreased. So the force might have decreased for the same amount of extension. But it could also be that the amount of force being transmitted through the ties had changed due to the re-orientation of the ties and struts. Fair enough. But, then I noticed that the rest length initially wasn't the same as the one specified via the initial nodal positions. 
Node 3 = (0, 0,10);
Node 4 =  (-5,10,0);
So, the initial rest length should be 15. But it's not. It's around 10.07. Why this discrepancy? Or is that not how the initial rest-length is decided? Does that mean 'AL' in the log file does not mean 'Actual length', i.e. 'Actual Nodal Distance'? 
Moreover, I had set the stiffness value at 100000 in 'actuatorConfig', but the initial stiffness is being shown as (after calculation, see file below) as 97485. Why is that?
I'm doing all this to make sure that the software is giving me correct results. If you have some sort of benchmark that you used to validate your code, could you direct me to it?

brtietz, 2015-10-07:
The actual length is the actual distance between nodes. Rest length will
change based on pretension.

We assume that the system is a linear spring with a cable attached to it (F
= k * dx + bv), which is true for most of our robots. dx is the difference
between rest length and actual length. The cable is assumed to be stiff in
tension. Therefore, we assume constant, linear stiffness. See the attached
figure for a visual example.

If you increase the rest length, RL is closer to AL, and force will
decrease since dx is smaller - excluding any effects of other rods.

As far as benchmarking - the files in test_integration are what we use. The
simplest one for this case is the TimestepIndependence test, which is based
on src/dev/btietz/timestepTest In this case, you could turn off the motor
and make sure gravity stretches the spring appropriately.

chiku00, 2015-10-07:
Well, I came up with an answer to my question as I was going to sleep. Since a considerable length of the string is usually kept inside the spool connected to a motor, when a Tension is applied at the free end, (neglecting inter-thread friction) the entire length of the string experiences the same tension. So, in a way, the un stretched length never changed. How is this for an explanation?

brtietz, 2015-10-07:
That's accurate. Typically he unstretched length only changes if the motor
moves. It is possible to configure tgKinematicMotor such that this is not
true, but I don't think anyone has done it.