NASA-Tensegrity-Robotics-Toolkit/NTRTsim #181: Adding a btHingeConstraint to a tgStructure

AcexSpade, 2015-11-02:
Hello,

I am working for the BEST lab on a gimbal system, and for that purpose I need to use constrained rigid bodies. I do not find a way to retrieve the btRigidBody that are linked with the rods added by addPair on the tgStructure. 

Any help would be highly appreciated !

Julien

axydes, 2015-11-02:
I'm not sure if a better solution has been developed but for my masters
thesis I linked rods together using the btHingeConstraint. You can use that
as an example or starting point.

See tgRodHinge and tgRodHingeInfo on this branch and directory:
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/DuCTT/src/dev/axydes/DuCTT/robot

AcexSpade, 2015-11-02:
Thank you very much for your answer. It might be what I'm looking for ! Do you have a file where these objects are implemented ?

Julien

axydes, 2015-11-02:
Look at the DuCTTRobotModel files in the same directory. Also, looks like I
was using tgDuCTTHinge not tgRodHinge, so I'd look at those files in the
same directory.

ryan1248, 2015-11-02:
Hi Julien,

Have you taken a look at Alexander's tgRodHinge* classes? Would that work
for what you need, or are there other constraints you're interested in?

NTRT Devs: it might be worthwhile to consider merging the tgRodHinge*
classes into NTRT -- thoughts?

-Ryan

ryan1248, 2015-11-02:
Link to Alexander's DuCTT files containing the tgRodHinge* classes:
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/DuCTT/src/dev/axydes/DuCTT/robot

-Ryan

AcexSpade, 2015-12-05:
Hi,

Sorry for the late reply, but I have managed to make my hinge using the sources you have provided.

Thanks a lot for that !

Julien

closed 2016-08-07 by apsabelhaus