NASA-Tensegrity-Robotics-Toolkit/NTRTsim #191: The order in which elements are specified has an effect on the nature of the resulting structure

ryan1248, 2016-04-03:
The order in which builders, pairs, and nodes seems to have a material effect on the resulting structure as noticed by the failure of the ICRA2015 integration test. 

The order of definition of pairs and builders (as seen in src/dev/btietz/TetraSpineStatic/TetraSpineStaticModel_hf.cpp) should not affect the resulting structure in any way. However, the error values from the ICRA2015 test are different in the following cases:
- The order of builder definition is changed (TetraSpineStaticModel_hf.cpp, lines 290-292)
- The order of builder application is changed, as in the new code from Queens

The effect (ICRA2015 test failure due to error values exceeded) can be replicated in the original code by changing the ordering of the builders in the test, or by applying the new code. 

Relevant files:
- tgcreator/tgStructureInfo.cpp -- this was changed from a builder iteration to a node/pairs iteration to ensure that each node/pair is only built by one builder
- src/dev/btietz/TetraSpineStatic/TetraSpineStaticModel_hf.cpp
- test_integration/CMakeLists.txt -- enable/disable the tests as needed
- test_integration/ICRA2015Tests/ICRA2015_test.cpp

simonkotwicz, 2016-04-05:
The reason the test results were different is not because the structure itself was different but because the order of strings was different (using the updated builder tools) and the test checked the tension level of the strings by index rather than by tag. I think this issue can be closed.

ryan1248, 2016-04-05:
Ah that's great to hear! Let's go ahead and close it out if we're confident
in the revised test.

-Ryan

closed 2017-02-08 by brtietz