NASA-Tensegrity-Robotics-Toolkit/NTRTsim #194: Error when I run build.sh

CarlosManriquez, 2016-08-04:
Hi
I have been tried to install the new NTRTsim-master toolkit, however, I got an error during the run build.sh part. The software display the error:

-- Configuring incomplete, errors occurred!
See also "/home/tanque/Documentos/NTRT/NTRTsim/build/CMakeFiles/CMakeOutput.log".
- ERROR: CMake for Bullet Physics failed.

I already checked the  CmakeOutput.log file but, honestly, I did not understand much of the file. I really need some help.

[image]

apsabelhaus, 2016-08-07:
Hi Carlos,

Could you describe the problem a bit more, and upload these log files so we can see them? Pastebin (http://pastebin.com/) is a good place to put files like this so that others can read and see them easily. Put the CMakeOutput.log file there, and provide a link to it.

It would also be useful if you put all the output from build.sh into a file for us to look at too. One way to save the output from a command into a text file is to use the >> operator on the command line, like:

./build.sh >> build_log_carlos_2016-08-XX.log

Also, did you complete all the steps in the INSTALL file? (https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/master/INSTALL). If one of these steps failed, which one, and what error message did it produce? What is the output when you run ./setup.sh in the top-most folder?

We'll get you up and running soon! :)

CarlosManriquez, 2016-08-08:
Hi apsabellhaus

Thank you for your help, I completed all the steps in the install file and aparently my computer did not have install the next packages

libglib2.0-dev
curl
freeglut3
freeglut3-dev
build-essential

So after installation, I re run the setup.sh and build.sh file and it works just fine.

apsabelhaus, 2016-08-12:
Hi Carlos,

The error you received about YAML::BadFile would not seem to be related to your original error. Do you still get the error with CMake and Bullet Physics? Again, we can help more if we can see the full log file. 

The new error you received is actually related to the YAML builder not being able to find the file you specify. (We've recently added a better error message about this.) You've slightly misspelled the path to the YAML file. It should be:

build/yamlbuilder/BuildModel resources/YamlStructures/BaseStructures/3Prism.yaml 

and not 

build/yamlbuilder/BuildModel resources/YAMLStructures/BaseStructures/3Prism.yaml 

I just pushed the new code that gives a better error. If you run 'git pull' from the NTRTsim directory, then run bin/build.sh again, and then run uild/yamlbuilder/BuildModel resources/YAMLStructures/BaseStructures/3Prism.yaml, you'll see the new error.

Hopefully this helps. And if you're planning to use NTRT, please sign up for our mailing lists to stay updated about progress on the simulator!

CarlosManriquez, 2016-08-18:
Hi apsabellhaus

The second error was not related to the first one. The first one was fixed when I reran setup.sh. And the second, I wrote the path as you told me and everything was ok. Now NTRT works fine. Thank you for all your help.

MiracleCK, 2016-09-02:
Hi
I also got an error during run bin/build.sh
Before run build.sh,everything is all right when I run setup.sh
There have several errors displayed in screen:
[image]
*** [tgcreator/CMakeFiles/tgcreator.dir/tgStructure.cpp.o] error1
*** [tgcreator/CMakeFiles/tgcreator.dir/all] error2

apsabelhaus, 2016-09-21:
Hello @MiracleCK, thanks for using our software. Could you post your errors in English? That way we can help you more. Also, please include a full log file, not a screenshot, like I instructed to Carlos above.

apsabelhaus, 2017-02-07:
Closed due to inactivity. NTRT doesn't currently support other languages anyway.

closed 2017-02-07 by apsabelhaus