NASA-Tensegrity-Robotics-Toolkit/NTRTsim #195: Disable automatic anchor placement at the edges of rigid bodies

apsabelhaus, 2016-08-07:
Currently, connectors are moved away from their nodes automatically, and moved to the edge of a rigid body. 

For example, see the tgBoxAnchorDebug branch,  and the app in dev/apsabelhaus/tgBoxAnchorDebugDemo/AppTgBoxAnchorDebugDemo, as of commit 9a80587. The spring should connect the centers of the two rods (discs), but instead is moved to their outer edge.

There should be a way to disable this functionality so that cables can be placed precisely when needed. Also, this should be documented in our API properly.

Feedback needed:
- What classes/methods implement this automatic anchor movement?
- Where do folks think a "do-not-automatically-move-anchors" flag should go? In the config of the connector, or the config of the rigid body?

I have a deadline to get my force sensor working, for which I need to disable anchor movement, so feedback is much appreciated as soon as possible.

brtietz, 2016-08-08:
Drew,

Remember that cables placed at the center of a rod provide no rotation
constraint in NTRT, so that rods with those types of placements will rotate
uncontrollably. Not what we want for a default behavior.

That said, if you look at tgSpringCableActuator's config:
http://ntrt.perryb.ca/doxygen/structtg_spring_cable_actuator_1_1_config.html
You'll see a "rotation" parameter that can be used to specify where on the
outer circle a cable can be placed.

What I'd recommend doing, to facilitate precise placement, is adding a
"radius" parameter to that config, and using it to control how close to the
center the cable is placed. 0 for dead center, and 1 for outside
(maintaining the current default).

As far as the current behavior, have a look at these:
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/bb272ea71e26d00468e02202bc64dcd133ceab02/src/tgcreator/tgBasicActuatorInfo.cpp
tgBasicActuatorInfo::createTgBulletSpringCable()
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/bb272ea71e26d00468e02202bc64dcd133ceab02/src/tgcreator/tgRodInfo.cpp

Brian

apsabelhaus, 2016-08-12:
Thanks Brian! The motivation makes sense, for sure. Free-spinning rods should not occur. (Also, I had no idea that's what "rotation" meant!)

As of commit 2695c90, I've provided a boolean inside the config of tgBasicActuator that can disable cable anchor movement. That seemed a bit easier than trying to add a radius: since I'm trying to use this with tgBox, I can't assume a circular face, and writing functions to do the math for different shapes seemed inefficient if I was going to set radius=0 anyway!. The default is still cable anchor placement at the edge. There are ample warnings in the tgBasicActuator file about using this flag, too.

Brian and others, let me know if you think this solution should be changed. I've written everything such that no one else's code should be effected (knock on wood.) 

Also, if folks have thoughts about the naming of this boolean flag, please let me know or just go ahead and make the change. The best I could think of was moveCablePointToEdge = true (default). 

I'm going to close this issue now, but do re-open if a problem arises.

closed 2016-08-12 by apsabelhaus

gtournois, 2016-12-05:
Hi Drew, 

I have been using this solution for my particular problem. But now I have a problem. My tensegrity model is standing on 4 rods and due to the free-spinning rods the model is like "floating" away. Do you have any idea how I could fix this?

Regards,
Guido