NASA-Tensegrity-Robotics-Toolkit/NTRTsim #196: tgBoxInfo calculates and incorrect mass, tgBox takes half-extents not full-extents

apsabelhaus, 2016-09-02:
The current call to the Bullet Physics function that creates a box, line 109 in tgBoxInfo.cpp, passes in the full width and height of a box:

 m_collisionShape = new btBoxShape(btVector3(width, length / 2.0, height));

However, the Bullet API says that the btBoxShape constructor takes "half extents", which would be half the width and half the height of the box. As a consequence, our boxes in Bullet are twice as tall and thick as they are according to NTRT. In other words, our constructor for tgBox does not actually take the full width and height, but only _half_ the width and height.

Changing this behavior would drastically affect other folks' code, so it does not make sense to me to fix this issue right now.

However, this does cause one major inaccuracy in our simulations: the getMass() function in tgBoxInfo does not calculate the correct mass, since it assumes the volume is (length x width x height) whereas volume is really ( length x width/2 x height/2). 

I plan to correct the getMass() function. Please let me know ASAP if your code depends on the mass of a tgBox.

brtietz, 2016-11-20:
For what its worth, this was already reported in #123

apsabelhaus, 2017-01-04:
Thanks Brian, I didn't see that issue. Since no one else seems to care much, I went ahead and implemented the fix - mass is now correct as of commit c4a4c92.

Since we've already been using width and height in this inconsistent way, it would make sense to keep that convention, right? We can just be very clear in the documentation that "length" is the full length, but "width" and "height" are half the width and half the height respectively.

closed 2017-01-04 by apsabelhaus