NASA-Tensegrity-Robotics-Toolkit/NTRTsim #199: Compression Joints in NTRT

51stpage, 2016-12-12:
I was wondering if there's a way to do end to end compression joints using two rods in NTRT?  I think I have a way to fake it by using a complicated multi-rod, multi-string approach similar to this [spine](https://i.ytimg.com/vi/71s-hMo-6vE/maxresdefault.jpg) but would prefer not to include the added complixity in the model if I can help it.  If not, I'll use the more complicated way.

vsunspiral, 2016-12-13:
can you explain what you mean by an end-to-end compression joint?
do you mean that you want to put a rotary joint between two rods?
Or something else?

vytas

> I was wondering if there's a way to do end to end compression joints using two rods in NTRT? I think I have a way to fake it by using a complicated multi-rod, multi-string approach similar to this [...]

-------------------------------------------------
Vytas SunSpiral    
Dynamic Tensegrity Robotics Lab
cell-        510-847-4600
Office:    650-604-4363
N269     Rm. 100

Stinger Ghaffarian Technologies
Intelligent Robotics Group
NASA Ames Research Center

I will not tiptoe cautiously through life only to
         arrive safely at death.

51stpage, 2016-12-13:
Sure.  I'm looking to model human hand-like structures so I'll have bones butting up against each other then rotating around an axis.  I've found references to 'rotation:' in the documentation though I haven't had a chance to explore how those exactly work.

If there's a way to do hinge and ball-joints that would be great.  As a side-query, is there a way to add arbitrary shaped objects in the world for the model to interact with?  I've seen what look like static ground features but was wondering about spheres and cylinders.

Thanks for getting back to me.