NASA-Tensegrity-Robotics-Toolkit/NTRTsim #210: Error when installing controller

wildsky95, 2017-04-03:
When i install a controller to a model i get the error below
How can i fix it
[image]

apsabelhaus, 2017-04-03:
Hi wildsky,

Your image is blurry and I can't see what is going on. Please take a screenshot and upload that instead so we can help you!

Also, controllers aren't "installed" in NTRTsim. Please explain exactly what you mean, and exactly what you did, then we can help you figure it out.

Welcome to NTRT.
-Drew

wildsky95, 2017-04-03:
thanks for your attention 
i coppied supperball example to the folder which i made in dev directrory then i deleted the supperball from the examples cmake file and add my folder to dev folders cmake list.after i added the controller on the Appsuperball and also i add the controller name to the executable files in cmake at mesioned directory then i wanted to compile the files by build.sh and i get this error which i attach the picture.
im researching on tensegrity models ill be grateful for your help
[image]
[image]
and also i cant run some of the executable files in build directory i run some of them and nothing happens,no simulation comes up ...!

PerryBhandal, 2017-04-04:
I'm assuming you copied SUPERball from examples into your dev folder. Is
that correct? If you did, did you just copy some of the files but not
subfolders (such as controllers)?

PerryBhandal, 2017-04-04:
Just saw the rest of your e-mail's body where you addressed part of my
question.

My question, restated: Did you copy the entire SUPERball folder (including
all subfolders) into /dev/amir?

wildsky95, 2017-04-04:
Yes i coppies all and i also deleted the supperball from its cmake that it wont compile an just amir folder to be compiled because of the name issues i think,i just want to try adding some controllers to then add mine and work on it but i crash into this error

PerryBhandal, 2017-04-04:
There's some change you're making beyond moving SUPERball to /dev/amir and updating the CMakeList which is causing your problem. Admittedly, that on its own isn't terribly helpful, but without seeing your changes it's difficult to specify what the exact issue is.

We can provide you more help if you can show us the exact changes you've made. The easiest way to do that is to go into your NTRTsim directory and run **git diff > changes.txt** then upload changes.txt.

wildsky95, 2017-04-05:
sorry for the delay 
i dont understand what you mean by uploading changes.txt 
i want to mention this
i also tried the ready controllers which were in the Appsuperrball without any changes only i deleted // but i get the same error.im really stuck on this please help 
thanks to your attention

wildsky95, 2017-04-05:
[image]
[image]
[image]
i uploaded the screenshots i made the changes which are just mentioned in the files itself why still i get this error