NASA-Tensegrity-Robotics-Toolkit/NTRTsim #217: Neural Net initialization failed

franciscocarreno, 2017-07-17:
During installation I got an error. The path to download nnImplementationV2.zip from dropbox doesn't exits anymore.

Error:
======================================================================
Initializing Neural Net...
Downloading nnImplementationV2.zip to /home/francisco/Documents/NTRTsim-1.1/env/downloads/nnImplementationV2.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   406    0   406    0     0     69      0 --:--:--  0:00:05 --:--:--    94
  0     0    0   408    0     0     61      0 --:--:--  0:00:06 --:--:--  1369
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
======== DOWNLOAD FAILURE =========
Encountered a failure while downloading:

Target URL: http://dl.dropbox.com/u/18209754/Blog/nnImplementationV2.zip
Local URL: /home/francisco/Documents/NTRTsim-1.1/env/downloads/nnImplementationV2.zip

Exiting now.
======== DOWNLOAD FAILURE =========
Neural Net initialization failed -- exiting now.
==========================================================================

PerryBhandal, 2017-07-17:
Can you run git log inside your NTRTsim folder and post the results?

I'm assuming you're using something other than the master branch, as the master branch was updated a while ago to pull deps from http://ntrt.perryb.ca

In the meantime you can manually resolve this by doing a find and replace

Replace

NEURALNET_URL="http://dl.dropbox.com/u/18209754/Blog/nnImplementationV2.zip"

with

NEURALNET_URL="http://ntrt.perryb.ca/storage/dependencies/nnImplementationV2.zip"

PerryBhandal, 2017-07-17:
Ahh, I see, presumably you're pulling that from the V1.1 tag on Github? Or maybe you downloaded that version directly from the NTRT site at nasa.gov?

In any case, you'll want to use the master branch version of NTRTsim. Version 1.1 is very old now, its last commit was on January 9th of 2015. Much has changed since then. 

Here is a direct link to get the current master branch as a zip:

https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/archive/master.zip

PerryBhandal, 2017-07-17:
> On Mon, Jul 17, 2017 at 10:14 AM, franciscocarreno <

Ahh, I see, presumably you're pulling that from the V1.1 tag on Github? Or
maybe you downloaded that version directly from the NTRT site at nasa.gov?

In any case, you'll want to use the master branch version of NTRTsim.
Version 1.1 is very old now, its last commit was on January 9th of 2015.
Much has changed since then.

Here is a direct link to get the current master branch as a zip:

https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/archive/master.zip

Perry

PerryBhandal, 2017-07-17:
Can you run *git log* inside your NTRTsim folder and post the results?

I'm assuming you're using something other than the master branch, as the
master branch was updated a while ago to pull deps from
http://ntrt.perryb.ca

In the meantime you can manually resolve this by doing a find and replace

Replace

NEURALNET_URL="http://dl.dropbox.com/u/18209754/Blog/nnImplementationV2.zip"

with

NEURALNET_URL="
http://ntrt.perryb.ca/storage/dependencies/nnImplementationV2.zip"

franciscocarreno, 2017-07-17:
I was using V1.1. Using current master branch now!
I run setup again and got: "Boost initialization failed"

git log:
francisco@ubuntu:~/Documents/NTRTsim-master$ git log
fatal: Not a git repository (or any of the parent directories): .git

[Boost initialization failed.txt](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/files/1153361/Boost.initialization.failed.txt)

PerryBhandal, 2017-07-17:
That is a known issue. Just run setup again (don't delete your existing
install or anything) and it shouldn't give you that error the second time.

franciscocarreno, 2017-07-17:
setup.sh finished successfully after second run. Thank you PerryBhandal!
I got a new error while running build.sh which I posted as new issue (#218)

danibix95, 2019-09-11:
> Can you run git log inside your NTRTsim folder and post the results?
>
> I'm assuming you're using something other than the master branch, as the master branch was updated a while ago to pull deps from [...]

Since the http://ntrt.perryb.ca/ server is currently not reachable, is it safe to assume that the Neural Network dependency code is the same as the one reported in the source code [learning folder](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/master/src/learning)?