NASA-Tensegrity-Robotics-Toolkit/NTRTsim #218: ERROR: CMake for Bullet Physics failed

franciscocarreno, 2017-07-17:
Running bin/build.sh I got error:
ERROR: CMake for Bullet Physics failed

Previously I finished setup.sh successfully after corrections (issue #217)
CMakeOutPut.log: https://pastebin.com/8XkNvDvH
Command line error:
[build_log_francisco_20170717.txt](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/files/1153941/build_log_francisco_20170717.txt)

Please help

Psygohn, 2017-07-17:
That error may be caused by missing dependencies.

Take a look at the INSTALL file in the root of the Github repo, and install
any missing dependencies. Once you have all deps installed you should nuke
your existing NTRTsim install and compile a new version from scratch.

Let me know if that does/doesn't resolve your problems (or this specific
error, at the very least).

Perry

PerryBhandal, 2017-07-17:
That error may be caused by missing dependencies.

Take a look at the INSTALL file in the root of the Github repo, and install
any missing dependencies. Once you have all deps installed you should nuke
your existing NTRTsim install and compile a new version from scratch.

Let me know if that does/doesn't resolve your problems (or this specific
error, at the very least).

Perry

franciscocarreno, 2017-07-18:
I installed all dependencies, erase env and build, run again setup.sh, and build.sh
I finished all with no error messages. I guess it's installed now. (see log attached)

I'm following the tutorial now. I'm trying to use YAML. When I run BuildModel nothing happens. I also tried "build/yamlbuilder/BuildModel resources/YAMLStructures/BaseStructures/3Prism.yaml" and got 
message 'YAML::BadFile' (see log attached)

[YAML 20170718 fran 1.txt](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/files/1155683/YAML.20170718.fran.1.txt)
[Build 20170718 fran 1.txt](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/files/1155685/Build.20170718.fran.1.txt)

PerryBhandal, 2017-07-18:
While running the build does compile all simulator applications, you'll still need to explicitly run an application -- many are included in NTRT. For instance, if you want to run 3prism-serialize you would navigate to build/examples/3_prism_serialize and run AppPrismModel.

franciscocarreno, 2017-07-18:
I see. I tried the examples. They run no problem. How can I build my own structures? Which tutorial could I follow?

PerryBhandal, 2017-07-18:
See the repo README. Many details and links there.

apsabelhaus, 2017-07-18:
Francisco,

Check out the video tutorial in the README file. It shows you how to use
YAML files as well as traditional tgModel applications. Welcome to NTRT!

Drew

-- 
Drew (Andrew P.) Sabelhaus
Graduate Student Researcher | UC Berkeley | Mechanical Engineering
Berkeley Emergent Space Tensegrities (BEST) Lab
NASA Ames Research Center | Intelligent Robotics Group | Dynamic Tensegrity
Robotics Lab

My pronouns are he, him, his. (What's
<https://www.ccsu.edu/lgbt/files/PreferredGenderPronounsForFaculty.pdf>
this? <http://lgbt.williams.edu/resources/trans-resources/pronouns/> Why?
<https://www.aclunc.org/article/frequently-asked-questions-whats-pronoun>)
Phone: (301) 807-9842 | www.apsabelhaus.com
------------------------------------------------------------------------------------------

franciscocarreno, 2017-07-18:
thank you guys!

franciscocarreno, 2017-07-18:
So sorry to bother you guys again…

I tried "git pull" to download the examples and I get this:

francisco@ubuntu:~/Documents/NTRTsim-master$ git pull
fatal: Not a git repository (or any of the parent directories): .git

I also tried this:

francisco@ubuntu:~/Documents/NTRTsim-master$ git checkout NTRT_Intro
fatal: Not a git repository (or any of the parent directories): .git

I then try gitinit

francisco@ubuntu:~/Documents/NTRTsim-master$ git init
Initialized empty Git repository in /home/francisco/Documents/NTRTsim-master/.git/
francisco@ubuntu:~/Documents/NTRTsim-master$ git pull
fatal: No remote repository specified.  Please, specify either a URL or a
remote name from which new revisions should be fetched.

I'm not sure which URL is asking me for. Please help!

PerryBhandal, 2017-07-18:
If you downloaded the zip I linked earlier and extracted it somewhere it just includes the master branch, but not the settings for the specific git repository you pulled it from. For git pull to work you would have needed to clone the repository, rather than downloading it as a zip.

If the specifics above aren't something you're familiar with, don't worry about it. I believe that for everything you'll want to do you can simply ignore any git commands.

(I'm 99% certain that after sending this e-mail an example from the tutorials will come up where the above rule is incorrect. Fortunately I can blame that on Drew.)

apsabelhaus, 2017-07-18:
Hey Perry/Francisco,

Francisco, it would be a good idea to get familiar with git before using
NTRT. The branches are important. Maybe try a git tutorial, and create an
example repository for yourself, and get familiar with clone/push/pull.

You'll need to clone the repository, or use one of the pre-built virtual
machines, in order to switch to the NTRT_Intro branch. As discussed in the
video, you need to install the simulator (implication: clone or use the
virtual machine) before switching to the NTRT_Intro and following the
video. Downloading a .zip file does not contain all the branches, so you
won't be able to switch to the NTRT_Intro branch inside the zip file. You'd
have to download a separate zip for the other branch. But since you'll be
using  both, it's best to just clone and not have to deal with two
installations.

To sum up: your error message at this point is not about NTRT. It's about
using git. There are many other resources online to help with that. I'll
help on this list for NTRT-specific problems, if you encounter any.

Good luck,
Drew

-- 
Drew (Andrew P.) Sabelhaus
Graduate Student Researcher | UC Berkeley | Mechanical Engineering
Berkeley Emergent Space Tensegrities (BEST) Lab
NASA Ames Research Center | Intelligent Robotics Group | Dynamic Tensegrity
Robotics Lab

My pronouns are he, him, his. (What's
<https://www.ccsu.edu/lgbt/files/PreferredGenderPronounsForFaculty.pdf>
this? <http://lgbt.williams.edu/resources/trans-resources/pronouns/> Why?
<https://www.aclunc.org/article/frequently-asked-questions-whats-pronoun>)
Phone: (301) 807-9842 | www.apsabelhaus.com
------------------------------------------------------------------------------------------

franciscocarreno, 2017-07-19:
I found some instructions on how to clone the repository.
https://help.github.com/articles/cloning-a-repository/
Actually for the first exercise in the video I need to download some examples with git pull.
I'll try to do this by tomorrow so I'm able follow the NTRT Introduction Seminar video.

jae-inkazama, 2020-05-15:
I was getting a similar error, and afterinstalling the dependencies and running setup.sh, the part where boost is doesnt progress. the timer keeps ticking along. Also Perry, your website where the virtual machine is also down.

PerryBhandal, 2020-05-16:
Hi Zain,
I should have posted about the site being down.

I'm moving the server to a new DC. Unfortunately it won't be back up until the 29th at the earliest.

Perry

jae-inkazama, 2020-05-16:
Hi Perry,

Oh damn that's bad.
I cloned the repostiories did as you said above about dependenciesand the part where boost is downloading seems to be stuck. It keeps ticking forever. Any leads,

Zain

PerryBhandal, 2020-05-16:
Yikes, completely forgot that the dependencies were hosted on my domain. I
figured only the buildbot would be affected which isn't super urgent.

I'll try to get it running again on a VPS asap. Hopefully it'll be up by
Monday. I'll send an e-mail once it's up.

jae-inkazama, 2020-05-16:
Thanks man! I'll remind you in the unlikely case it escaped your mind.

PerryBhandal, 2020-05-18:
Hey Zain,
Sorry, unfortunately it won't be done until Thursday night. It's a definite, though.

Perry

jae-inkazama, 2020-05-18:
Hm it's okay, I forked the legacy version as suggested by Massimo, it worked after throwing a boost initialisation failure the first time.
Cheers

jae-inkazama, 2020-05-18:
A little guidance on the data logging part would be great tho. I have copied the BuildTensegrity.cpp file for a yaml model and added the code in it. Do I need to make it somehow?

apsabelhaus, 2020-05-18:
Hi Zain et al.,

First - Zain, responding to your individual email to me, I would not
recommend actively using NTRT for something with a deadline like a class
project, much less waiting until one week before the deadline to get
logging started. NTRT is out of date and is not actively used by most
developers. For me in particular, NTRT isn't my research anymore, so all
code is offered as-is. That said, I have a moment, so here are some
pointers on logging / Cmake.

There were already some answers to your questions in various places, so be
sure to search around in the future in case folks take time to respond. In
particular,

(1) The failed boost make is a known problem. As it happens, NTRT only uses
boost libraries that successfully build, no reliance on the broken ones.
So, none of the developers have needed to fix this problem yet. Just run
the setup script twice (or more) until everything completes.

(2) Logging - see
https://ntrtsim.readthedocs.io/en/latest/data-management-and-logging.html.
It's linked in the README.

(3) Building new apps - yes, you will need to put a new CMakeLists file in
the directory you're using, or else your code won't be compiled. The video
tutorial (see the
https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/NTRT_Intro
branch) has a bit on this, but does not necessarily cover using new
controllers with YAML files, which I'm assuming is why you're copying
BuildTensegrity around. You can look at any of the applications in src/dev
for example CMakeLists files with new app/model/controller combinations,
the pattern should be relatively self-evident.

Best of luck,
Drew

Andrew P. Sabelhaus
Postdoctoral Research Associate | Soft Machines Lab
Carnegie Mellon University | Mechanical Engineering

My pronouns are he, him, his. (What's this? www.mypronouns.org)
Phone: (301) 807-9842 | www.apsabelhaus.com
----------------------------------------------------------------------------------------------

jvmccall, 2020-09-20:
I am having the same CMake for Bullet Physics failed error. 

I am using Ubuntu
I installed:
- g++ 7 (the compiler for cmake)
	- libglib2.0-dev
	- curl
	- freeglut3
	- freeglut3-dev (glut for bullet graphics)
	- cmake
	- build-essential
        - unzip
- python2.7

I download and unzip everything. I run setup twice until I get a message saying setup is complete. The first time I run setup it indicates boost initialization failed which the install instruction says is ok.

Then I run build and I get the CMake Bullet physics error. I have tried installing, and updating dependencies multiple times. I have tried Ubuntu 20 and 18.  I have deleted and unzipped new master folders a few times.

Build Error Pastebin
https://pastebin.com/0gnDsbbJ

Setup Complete message 
https://pastebin.com/sNuGL2QK

CMakeOutputLog
[CMakeOutput.log](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/files/5252181/CMakeOutput.log)

I am running on a raspberry pi4 4gb. I would try to use a virtualmachine on my main computer but the .ova link is not working.

I am not sure what else to try. Thank you for any advice you can give.

apsabelhaus, 2020-09-21:
@jvmccall The CMake output tells you the issue:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLIB_LIB_INCLUDE_DIR
```

There seems to be a problem with however glib is set up or configured on your computer. I personally have not seen this error before, so it's likely that the problem is unique to your setup as opposed to an issue with NTRT. I would recommend searching around on the internet for similar errors and trying out different solutions. If you figure out what needs to be added to the INSTALL instructions, let us know.

For some debugging to compare to a working NTRT installation:

The environment variable ```GLIB_LIB_INCLUDE_DIR``` does not seem to be defined, so you wouldn't need it to be either:

```
drew@drewbook-linux:~$ echo $GLIB_LIB_INCLUDE_DIR

drew@drewbook-linux:~$
```

But it does appear to be set via CMake internally somehow:

```
drew@drewbook-linux:~/repositories/NTRTsim$ grep -r -i GLIB_LIB_INCLUDE_DIR .
./src/inc.CMakeBullet.txt:    FIND_PATH(GLIB_LIB_INCLUDE_DIR glibconfig.h 
./src/inc.CMakeBullet.txt:                ${GLIB_LIB_INCLUDE_DIR})
./build/CMakeCache.txt:GLIB_LIB_INCLUDE_DIR:PATH=/usr/lib/x86_64-linux-gnu/glib-2.0/include
drew@drewbook-linux:~/repositories/NTRTsim$
```

My first suggestion is to check and see if the glib include directory is present on your computer in a similar enough place as mine, and if the glibconfig.h file is somewhere. If not, then you may not have actually installed everything you need. If so, then something may have gone wrong during your setup of CMake. Maybe try installing glib before cmake when ```apt install```-ing the dependencies.

jvmccall, 2020-09-21:
@apsabelhaus Thank you. I had seen that output and I tried installing libglib2.0 in addition to libglib2.0-dev in case that fixed the problem. 

I think I did install libglib2.0-dev before cmake. I have been installing the dependencies in the order listed in the INSTALL document.

I will check and see if it is in the right location and will post if I find out the solution. 

In the meantime, is there an active link for the .ova file so I can try setting up a virtualmachine for NTRT? The perry link is down.

apsabelhaus, 2020-09-21:
@jvmccall No idea about the virtual machine file. It's been many years since we used it and many years since Perry moved on from the project. I would recommend signing up for the ntrt listservs (see the official website for NTRT on the NASA page) and sending out an ask to current users to see if someone could upload a vm image for you somewhere.