NASA-Tensegrity-Robotics-Toolkit/NTRTsim #224: NTRTsim does not compile with gcc 7 / g++ 7

apsabelhaus, 2018-12-19:
As of Ubuntu 18 / Debian stretch, gcc and g++ version 7 are the default. NTRTsim does not compile with these versions, there is a change somewhere between 4.8 and 7 that broke something we relied upon.

A temporary fix is included in the INSTALL instructions to downgrade to 4.8 and have both compilers present on a system using Ubuntu's update-alternatives.

This error should be fixed, though. It may be related to ongoing issues with using C++98 or the (by now, very old) Bullet 2.82.

furlong-cmu, 2019-05-03:
I believe I've managed to fix the problems with the getline() > 0, and the problems with the static initialization in the yamlLoader.  But I think it makes the code not backwards compatible with older c++ versions (language not the compiler).  I'm pretty new to the code base though, so how do I test it to make sure I haven't goofed everything up?

apsabelhaus, 2019-05-03:
@furlong-cmu that's great news! Unfortunately, our testing suite is mostly nonexistent. Many people have dropped off working with NTRTsim (example: a student does a machine learning test, graduates, and leaves their code unmaintained) so we don't bother going back and re-writing people's work that will never be used again. So, it's not been worth anyone's time to integrate a full testing suite.

If I understand correctly, you're saying that your code has moved away from C++99. Is that right? Unfortunately, I believe the version of Bullet Physics we use adheres to C++99. I don't know if that breaks on upgrade or how, but I do seem to remember that sticking with this older version of Bullet is what's been holding back moving to e.g. C++11. To make sure your changes work in general with gcc7+ or whatever more recent C++ standard used now, I'd:

- clone a fresh version of NTRTsim into another folder,
- on master, patch it with the changes you made, 
- run the setup.sh script to recompile all libraries (including Bullet 2.7 or whatever) appropriately, with the same compiler / C++ standard you now use for NTRTsim
- If that runs (*NOTE: see another Github issue about a known problem with compiling boost), and running bin/build.sh compiles NTRTsim, then I'd say we're good to go.

A good way to double check that you haven't broken other people's work (e.g. even if it compiles) is to run all the apps in the 'example' folder with your code changes, and make sure they don't segfault.

I'm personally unlikely to use NTRTsim for my research moving forward (or possibly, ever again), so I do not recommend putting much effort into serious fixes here. If any researcher seriously wants to do tensegrity simulations in the future, I think Vytas' recommendation holds:

1) Fork NTRT. 
2) Hack out individual contributions from /dev/, and set the standard that no Apps are to be developed in the forked repository (make a framework for people to have their own individual repos for their own apps.)
3) Upgrade to use Bullet Physics 3+.
4) Upgrade to a newer C++ standard.