NASA-Tensegrity-Robotics-Toolkit/NTRTsim #226: Model Complex Structures

jae-inkazama, 2020-01-26:
I want to know how to model complex tensegrity structures such as the 30 strut icosahedron/ dodecahedron as a YAML file.

apsabelhaus, 2020-01-26:
Hi Zain,

I'm responding here to the email you sent me also. First, welcome to the NTRT community, I hope our software is helpful for you.

Second, an issue in Github is not the place for your question. The README.md for the repository (displayed at https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim) describes how to sign up for our mailing lists. That's where you should ask the question, since others besides myself can respond there. The NTRTsim software is relatively out-of-date by now, and there are few users left to discuss your models with you, so the email list may not be very active. It is still the best place, though.

Finally, to answer your question about the geometry and node positions of 30-struct tensegrities: if you can't find an example in the repository somewhere, and if there are no research papers with the geometry you desire, then it will be up to you as a research task to test and check coordinates for your new design. This is the design process - you will have to test and try yourself! In the past, various teams have tested geometries with sticks and strings in lab, then measured the coordinates, then used those in NTRT. Good luck with your research and design project.

closed 2020-01-26 by apsabelhaus

jae-inkazama, 2020-01-26:
Thanks for prompt reply!