NASA-Tensegrity-Robotics-Toolkit/NTRTsim #227: Trouble about cannot build a specific app

alwaysqi1121, 2020-04-09:
I have run the script of setup and build ,thenI can just run the examples in build/examples, but I cannot build a specific app ,and I cannot run BuildModel in NTRTsim-master/build/yamlbuilder. 
I beg your help!

alwaysqi1121, 2020-04-09:
And I tried to download the newest code form Github, when I run the script of build.sh,it turned out : ‘constexpr’ does not name a type,and ‘rodDensity’ was not declared in this scope,
and it ended with :
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
and I am sure about that that all of the dependencies are installed.

JEB12345, 2020-04-09:
Hello alwaysqi1121. I just compiled and built the code on master and was able to run the BuildModel in build/yamlbuilder.

It is hard to debug your issue without knowing the exact commands you used or any edits you may have made in your code. I would recommend you delete your NTRTsim folder and re-download the source code. Then follow the instructions in INSTALL. You will need to run setup.sh multiple times (probably 3 or more times) in order to get the setup process to finish. Once you get the setup complete message, you can run bin/build.sh. 

Once all that is complete, try to run the build/yamlbuilder/BuildModel app. Know that you need to specify the location of the yaml file in relation to your current directory or the full path.

alwaysqi1121, 2020-04-10:
Thanks for your reply,I have read the instruction in INSTALL,and firstly，I run  setup.sh, then  it showed setup complete,and guided me to use bin/build ,when I was running build.sh, it showed

alwaysqi1121, 2020-04-10:
Thanks for your reply !
I tried to download the newest codes from GitHub,firstly,I am sure bout that all of the dependencies in INSTALL were installed. then ,I run setup.sh several times and then it showed :set complete and guided me to run  bin/build.sh. Secondly，I run bin/build.sh.I just did these.But it failed,I want to know does it need any other dependencies?And my classmate have the same problem,too,we tried to solve the problem,but we failed .errors were like :‘constexpr’ does not name a type , ‘boxRollFriction’ was not declared in this scope,and [yamlbuilder/CMakeFiles/BuildModel.dir/all] Error 2.and, Makefile:83: recipe for target 'all' failed,
make: *** [all] Error 2

Sorry to trouble you again,but I have a question: can I just open the file in NTRTsim-master/resources/YamlStructures/BaseStructures and change figures in them to build a new model on the basis of the examples'figures?
Because I opened a terminal in NTRTsim folder which is previously installed,and typed build/yamlbuilder/BuildModel resources/YAMLStructures/BaseStructures/3Prism.yaml      but it turned out that :The YAML parser threw a BadFile exception when trying to load one of your YAML files. The path of the structure that the parser attempted to load is: 'resources/YAMLStructures/BaseStructures/3Prism.yaml'. 
Check to be sure that the file exists, and that you didn't spell the path name incorrectly.

It is my graduation design,and I am very interested in tensegrity robot.And I wonder if there are some other ways to simulate?
Thank you very much!

JEB12345, 2020-04-10:
We recently updated some of our code to be more C++11 compliant, which breaks the build for any g++ compiler less than 7. More specifically, the latest commit on master no longer compiles under g++ 4.8. This is a problem if you are using an older version of linux (e.g. Ubuntu 16), or if you followed the install instructions to use g++-4.8 under update-alternatives.

I would recommend you update g++ to version 7+. If that is not possible, I did create a new legacy branch called "legacy_gcc-4.8". That branch is the current version of master with the C++11 fixes reverted.

Note: If you update your g++ version, you will need to delete the env/ folder and re-run setup.sh. If you switch over to the legacy_gcc-4.8 branch, then you don't necessarily need re-run setup.sh unless you are starting from a new download from github.

As for your issue with the YAML example, it seems that you are not referencing the 3Prism.yaml file correctly (so the app can not find the file). Let's assume you are in the build/yamlbuilder/BuildModel folder. Then a command you can call to run the app is as follows:

```bash
$ ./BuildModel ../../resources/YamlStructures/BaseStructures/3Prism.yaml
```

If you are in the main folder NTRTsim, then the a command you can call is:

```bash
$ build/yamlbuilder/./BuildModel resources/YamlStructures/BaseStructures/3Prism.yaml
```

Notice that I am referencing the files relative to my current directory.

alwaysqi1121, 2020-04-11:
Thank you very much!
I tried the command you gave me ,and it worked.
Previously,I tried the code that provided in NASA Tensegrity Robotics Toolkit Tutorials (http://ntrtsim.readthedocs.org/en/latest/index.html),the command is at YAML Model Builder/Running Structures in the Simulator.However,it misses  the './' in the code .so ,I cannot open the app and it turned out errors.It is hard for a beginner.Thank you !
By the way, I have installed the NTRT on the  basis of previous code before ntrt.perryb.ca was no longer running and now it turns out that it works ,so,can I use this version of simulator for a very long time without update?

alwaysqi1121, 2020-04-11:
Thank you very much!
I tried the command you gave me ,and it worked.
Previously,I tried the code that provided in NASA Tensegrity Robotics Toolkit Tutorials (http://ntrtsim.readthedocs.org/en/latest/index.html)  ,the command is at YAML Model Builder/Running Structures in the Simulator.However,it misses the './' in the code .so ,I cannot open the app and it turned out errors.It is hard for a beginner.Thank you !
By the way, I have installed the NTRT on the basis of previous code before ntrt.perryb.ca was no longer running and now it turns out that it works ,so,can I use this version of simulator for a very long time without update?

alwaysqi1121, 2020-04-11:
I update g++ to version 7,and  it worked!
I use the version of Ubuntu 16.04,it turned out errors while building  because of the version of g++.
Thanks a lot!

closed 2020-04-13 by JEB12345