NASA-Tensegrity-Robotics-Toolkit/NTRTsim #228: How to get the model to move?

alwaysqi1121, 2020-04-27:
Sorry to trouble you!
I have read the document(https://ntrtsim.readthedocs.io/en/latest/), and the article I found in http://www.magicalrobot.org/NTRT/Docs/NTRTv1.1/html/index.html . So I have a preliminary understanding of it. And I watched the video that uploaded to YouTube. 
However, I can just creat a static model in NTRTsim now, I want to know how to get the model to move.  I don’t know what is the workflow of NTRTsim, when I want to get the model to move, what should I do besides creating a static model?To be exact, what is the relationship between the folders in NTRTsim and what is the workflow?
I tried to learn, but it seems hard for a beginner like me.I think if I konw the workflow of NTRTsim,it would be simpler.
Thank you!

hany606, 2020-04-30:
> I have read the document(https://ntrtsim.readthedocs.io/en/latest/), and the article I found in http://www.magicalrobot.org/NTRT/Docs/NTRTv1.1/html/index.html . So I have a preliminary understanding [...]

As I know, there are some controllers that you can use, you can find their header and source files in [the controllers directory](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/master/src/controllers). The controllers control the cable itself, in that case, you will be able to make the structure to move.

Moreover, [this is a very good example](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/tree/NTRT_Intro/src/dev/introSkeleton) to show how to create a length controller based on one of the basic controllers and how you can integrate it with the structure. Also, I believe [this video](https://www.youtube.com/watch?v=jyP5h_t73xw) is really helpful, most probably you saw it before.

The basic structure for the project folder for you could be:
1. Main file which will be "App<name>.cpp" e.g. "App3Bar.cpp"
2. Controller e.g. "LengthController.cpp , LengthController.h"
3. Structure e.g. "threeBarModel.cpp, threeBarModel.h"

To understand how the usage of the structure with the controller, you can find the order in the main file e.g. [here](https://github.com/NASA-Tensegrity-Robotics-Toolkit/NTRTsim/blob/2e568056dcbc5bd8273a3d9f53ff94e07fc290e0/src/dev/introSkeleton/App3BarTraditional.cpp#L69-L86)

apsabelhaus, 2020-05-18:
Closing because github issues are not the right place for how-to questions.

closed 2020-05-18 by apsabelhaus